g4klx/MMDVM #8: Received a NAK from the MMDVM, command = 0x11, reason = 5

iw9grl, 2016-05-15:
Hi

this problem appear during the DSTAR reception through XReflector .
It seems a not enough buffer space .

iw9grl, 2016-05-19:
There is not anyone else who has this kind of problem?

[image]

g7npw, 2016-05-19:
Yes I have exactly same problem on my system from DStar network transmissions. It has been reported in the yahoo group also.

Jonathan is a very busy chap so I'm just waiting patiently for a fix.

g7npw, 2016-05-19:
On 19/05/2016 14:39, iw9grl wrote:
>
> There is not anyone else who has this kind of problem?
>
> capture 
> <https://cloud.githubusercontent.com/assets/16877668/15395383/bcf81d28-1dd7-11e6-88c3-5a3f460f472d.JPG>
>
>
Even transmissions from the local machine time server cause it.

M: 2016-05-19 15:45:00.360 D-Star, received network header from GB7DC   
/TIME to CQCQCQ
W: 2016-05-19 15:45:01.170 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 15:45:01.180 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 15:45:01.196 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 15:45:01.202 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 15:45:01.233 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 15:45:02.798 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 15:45:02.819 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 15:45:02.840 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 15:45:02.861 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 15:45:02.893 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 15:45:02.914 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 15:45:02.935 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 15:45:02.956 Received a NAK from the MMDVM, command = 
0x11, reason = 5
M: 2016-05-19 15:45:05.582 D-Star, received network end of transmission, 
5.2 seconds, 0% packet loss, BER: 0.0%
M: 2016-05-19 16:00:00.077 D-Star, received network header from GB7DC   
/TIME to CQCQCQ
W: 2016-05-19 16:00:00.893 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 16:00:00.899 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 16:00:00.909 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 16:00:00.920 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 16:00:00.951 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 16:00:00.972 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 16:00:00.993 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 16:00:01.551 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 16:00:01.577 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 16:00:01.598 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 16:00:01.614 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 16:00:01.636 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-19 16:00:01.657 Received a NAK from the MMDVM, command = 
0x11, reason = 5
M: 2016-05-19 16:00:04.198 D-Star, received network end of transmission, 
4.1 seconds, 0% packet loss, BER: 0.0%

Cheers Dom G7NPW

iw9grl, 2016-05-19:
Hi Dom ....

no problem, I hope that Jonathan know this kind of issue .... I can wait .... thank you
73 de IW9GRL

g7npw, 2016-05-22:
I just downloaded and tried the latest MMDVMHost b16fe29 and can confirm 
this problem seems to be gone now

Dom G7NPW

iw9grl, 2016-05-22:
I just upgraded seems to work .... I leave it in test ....

73

iw9grl, 2016-05-22:
don't work..... the problem is still present

g7npw, 2016-05-22:
I will investigate further on my setup. Was persistently doing it on every DStar transmission from the network but when I upgraded earlier it did not do it once. 

I shall log into a busy reflector and then review my logs.

iw9grl, 2016-05-23:
I noticed that, I receive the NAK in particular on a trasmission of a single ham operator !!!!!
Could you try, if possible, to receive on XRF 911A, when transmit IW5ECS, he using a IC92 Icom transceiver......

iw9grl, 2016-05-24:
[image]

g7npw, 2016-05-24:
I can confirm I am also seeing some errors from DStar Network. Although 
not as often.

M: 2016-05-24 08:04:02.199 D-Star, received network header from GW0JTU  
/7100 to CQCQCQ   via DCS005 B
W: 2016-05-24 08:04:03.066 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-24 08:04:04.197 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-24 08:04:04.218 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-24 08:04:04.239 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-24 08:04:04.260 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-24 08:04:04.281 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-24 08:04:04.307 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-24 08:04:04.328 Received a NAK from the MMDVM, command = 
0x11, reason = 5
W: 2016-05-24 08:04:04.364 Received a NAK from the MMDVM, command = 
0x11, reason = 5
M: 2016-05-24 08:04:11.519 D-Star, received network end of transmission, 
9.7 seconds, 0% packet loss, BER: 0.0%

73 Dom G7NPW

iw9grl, 2016-06-07:
any updates ??

73

g7npw, 2016-06-07:
On 07/06/2016 15:16, iw9grl wrote:
>
> any updates ??
>
>

Checked that earlier and still doing it on my system when connected to 
DCS005Z doing a quick echo test earlier. Also a few drop outs and beeps 
on the returned audio :(

Cheers Dom

iw9grl, 2016-06-07:
Same things Dom ....... we can only wait ........... 73

g4klx, 2016-07-20:
Try the latest host and see what happens.

iw9grl, 2016-07-20:
Hi Jonathan ..... 
I'm compiling the last host in this moment, I leave everything in tests and in the next few days I can provide if needed, a  log file to you .
Thank you so much
73 de IW9GRL

closed 2016-07-20 by g4klx