g4klx/MMDVM #16: EXTERNAL CLOCK 19.2Mhz Problem.

ik6qoo, 2016-08-05:
I'm compiled the last version of MMDVM with C4FM / DMR Filter and when i enable the external clock the led of TX is alwais ON and the modem not decode .
With internal clock is all ok .
I have not tried other clock ( 12 or 14.4 ) but I am sure that my clock 19.2 is mounted properly and running.
What could be the problem ?
is an error code ??

73s de IK6QOO Gianluca

iw9grl, 2016-08-05:
Gianluca, prova a premere il pulsante di reset dopo aver alimentato arduino dall'usb.... Se funziona ho la soluzione.... 73

ik6qoo, 2016-08-08:
Grazie del suggerimento  , domani  ricarico il firmware per il TXCO esterno e provo nuovamente .

ik6qoo, 2016-08-09:
Provato il reset ma non funziona .. Ho misurato con un frequenzimetro il pin di arduino dove è collegato il TXCO e leggo perfettamente 19,2Mhz .

iw9grl, 2016-08-09:
Stai alimentando il quarzo dai 3.3v di arduino o stai regolando i 5v???  Io avevo un drop sui 3.3v col quarzo collegato e non funzionava nulla, alimentando il quarzo dai 5v con un 78l33 tutto ha funzionato. 73

ik6qoo, 2016-08-09:
Ho un regolatore da 3.3 collegato ai 5V di arduino ( schema SP8NHT) , dici di prendere i 3,3 direttamente da arduino ?

iw9grl, 2016-08-09:
No non è necessario.... A questo punto non so cosa possa essere, mi spiace

ik6qoo, 2016-08-09:
Secondo me è un errore dell’attuale firmaware  , aspetto nuovi sviluppi . Grazie  comunque !

closed 2016-10-03 by g4klx

ik6qoo, 2016-10-03:
Risolto con external clock 12mhz tutto OK .

vk4tux, 2016-10-03:
The clip sine wave txco is not a good design, Use HCMOS > 12.0000 MHz ;

bridge connections >  R2, R26, C22    |   IC4 not installed, R20 not 
installed & not bridged. (C20 & others pending install)

http://vk4tux.duckdns.org/6.1/20160818_154426.jpg

ik6qoo, 2016-10-04:
solved with external clock 12MHz .

TNX

vk4tux, 2016-10-04:
19.2 or 14.4 or 12.288 work just as well.