g4klx/MMDVM #18: No A0 decoding on SP8NTH option

vk4tux, 2016-08-25:
In latest SP8NTH FW ; hardware & RX audio has been thoroughly checked through to A0, however no rx signal is detected, shown or indicated by receive LED. (A11), so unusable at the moment.

http://vk4tux.duckdns.org/6.1/6.1_rx.png

TX radio side works fine.

config.h

 *   Copyright (C) 2015,2016 by Jonathan Naylor G4KLX
 *
 *   This program is free software; you can redistribute it and/or modify
 *   it under the terms of the GNU General Public License as published by
 *   the Free Software Foundation; either version 2 of the License, or
 *   (at your option) any later version.
 *
 *   This program is distributed in the hope that it will be useful,
 *   but WITHOUT ANY WARRANTY; without even the implied warranty of
 *   MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 *   GNU General Public License for more details.
 *
 *   You should have received a copy of the GNU General Public License
 *   along with this program; if not, write to the Free Software
 *   Foundation, Inc., 675 Mass Ave, Cambridge, MA 02139, USA.
 */

#if !defined(CONFIG_H)
#define  CONFIG_H

// Allow for the use of high quality external clock oscillators
// The number is the frequency of the oscillator in Hertz.
// For 12 MHz
// #define EXTERNAL_OSC 12000000
// For 14.4 MHz
// #define EXTERNAL_OSC 14400000
// For 19.2 MHz
// #define EXTERNAL_OSC 19200000

// Allow the use of the COS line to lockout the modem
// #define USE_COS_AS_LOCKOUT

// Use pins to output the current mode
// #define ARDUINO_MODE_PINS

// For the original Arduino Due pin layout
// #define  ARDUINO_DUE_PAPA

// For the ZUM Board V1.0 pin layout
// #define  ARDUINO_DUE_ZUM_V10

// For the SP8NTH board
#define  ARDUINO_DUE_NTH

// To use wider C4FSK filters for DMR and System Fusion on transmit
// #define WIDE_C4FSK_FILTERS_TX
// To use wider C4FSK filters for DMR and System Fusion on receive
// #define WIDE_C4FSK_FILTERS_RX

// Pass RSSI information to the host
// #define SEND_RSSI_DATA

#endif

If no issue is apparent, can a debug be added to monitor A0 > log please?

g4klx, 2016-08-25:
The IO for the NTH board hasn't changed for a long time and no-one else has complained about it not working. Can I suggest you ask on the group about it? It may be an idea to switch the RSSI option off as it's known not to work on the non rssi branch anyway.

vk4tux, 2016-08-25:
It is now working , incl with rssi and PWM options. On latest FW and 
host. I had to modify the SP8NTH V6.1 hardware design to work on 
12.5KHz, i.e Taits TM8110's which I use.

It is all good thankyou. There was no issue with the FW of course.

closed 2016-10-03 by g4klx