g4klx/MMDVM #22: GPLv2 and non-commercial clause

daniestevez, 2016-10-04:
I've seen the following in the project's README:

"This software is licenced under the GPL v2 and is intended for amateur and educational use only. Use of this software for commercial purposes is strictly forbidden."

I believe that restricting the software to amateur and educational use only is incompatible with the GPLv2.

There is the same problem in MMDVHost and YSFClients.

g4klx, 2016-10-04:
It is not a problem. I am not going to explain it in public, but without the non-commercial clause I would be forced to delete the complete project.

If you don't like it, please don't use it.

closed 2016-10-04 by g4klx

daniestevez, 2016-10-04:
The only reason I contacted you here is because I couldn't find your email address on qrz.com or on github.

It's not about whether I like your software or not or about whether I use it or not, and it's a problem indeed. I've also seen you're using some code from Robert Morelos-Zaragoza's book, whose licence is incompatible with the GPLv2.

I would be happy to continue this discussion by email. You can find my address on github.

andyn, 2017-05-11:
>It is not a problem.

It isn't for people who are using your code, but it might be a legal problem for you.

GPL does not allow limiting the use of the software for any purpose. People are free to use your code for any purpose they want and that noncommercial clause in unenforceable, as the GPL explicitly says that any and all additional restrictions are void.

>I am not going to explain it in public, but without the non-commercial clause I would be forced to delete the complete project.

You're basically admitting that you have incorporated encumbered source code into your project.

If it's a copyright licensing problem, you have taken other people's code and relicensed it without permit.

If it's a patent licensing problem, it's up to the commercial users to get the required licenses. The issue is completely orthogonal to the concept of copyright. If it's about patents owned by you or your employer, you're not giving any implicit patent grants since the code is licensed under GPLv2, not version 3.

If you just want to keep the copyright to yourself in the case you want the exclusive rights to sell a software product in the future, there's nothing that prevents you from changing the license for all subsequent releases. Note that changing the license at that stage will not be possible unless you remove all contributed code or get a copyright waiver from the contributors.