g4klx/MMDVM #24: Can't use TCXO with odd frequency

reiser4, 2016-10-25:
(Repost from MMDVMHost: I think this repo is more appropriate)

Hello, I'm trying to use a TCXO at 11.289600 mhz.
Decode stops after about 0.7 seconds and I can't improve the situation acting on the OscOffset parameter.
I guess the problem is the integer division by 24000 and 48000.
Could MMDVM support my TCXO or should I throw it in the garbage and get a different frequency?
thanks.

phl0, 2016-10-25:
As IW9GRL already commented in https://github.com/g4klx/MMDVMHost/issues/183 the TCXO frequency has to be divisible by 24000 which 11289600 is clearly not. So this TCXO cannot be used with MMDVM.

reiser4, 2016-10-25:
Hi, I got that, but why? Can't I set is as 11280000 and adjust for +851ppm error?

g4klx, 2016-10-25:
No, it needs to be divided by the CPU. Please get a suitable TCXO, they're not expensive.

reiser4, 2016-10-26:
Okay, thank you. I suggest to add a note in the config file, saying that only those values are supported and other values won't work at all.

phl0, 2016-10-26:
Should be sufficient now I guess? https://github.com/g4klx/MMDVM/pull/27

phl0, 2016-10-26:
Ok, its in the Config.h now.

vk4tux, 2016-10-26:
If you look at config.h , that's why they are the option's given, the 
exception being 12.288 which also works.

closed 2016-10-31 by g4klx