g4klx/MMDVM #30: ARDUINO_DUE_PAPA section, pin missing

ik1whn, 2016-11-04:
File IODue.cpp, section:
 #if defined(ARDUINO_DUE_PAPA)

these lines are missing:
#define PIN_DSTAR           9
#define PIN_DMR             8
#define PIN_YSF               7
#define PIN_P25               6

phl0, 2016-11-04:
Pins 7 and 8 will collide with what is already defined:
```
#define PIN_COS                7
#define PIN_PTT                8
```

ik1whn, 2016-11-04:
Ok but without defining the pin arduino ide exit with error
exit status 1
'PIN_DSTAR' was not declared in this scope

phl0, 2016-11-04:
Yes sure. But the question is which pins to use for these. I don't have the Papa board pinout to see which ones would be free to use. Is there a schematic or pic from the papa board?

ik1whn, 2016-11-04:
You can use pin 4 and 5 instead of 7 and 8
pin 3, 4, 5, 6 and 9 are free

phl0, 2016-11-05:
I made a change based on the pin numbers you mentioned. Could you please give the code from my dev branch a try: https://github.com/phl0/MMDVM/tree/fixPAPA?files=1
If that works correctly I will make a pull request to the official code.

phl0, 2016-11-05:
Btw: Could you make a photo of the PAPA board? Did a Google search yesterday but could not find anything ... Would be interested in the hardware.

ik1whn, 2016-11-05:
Thanks Florian,
I try it probably tomorrow, today I'm gone.
The Papa board is the firts board for mmdvm, It is a prototype derived from
a gmsk d-star board realized by Jim ki6zum.
Here  someone still uses it but rarely update the system.
I send you a photo and schematic of the modification of original gmsk board.

73 Mauri ik1whn

phl0, 2016-11-05:
mni tnx :-)

ik1whn, 2016-11-06:
HI Florian,
the code works correctly, many thanks

Mauri ik1whn

phl0, 2016-11-06:
Die you actually test the mode pins? Or did you just compile it?

ik1whn, 2016-11-06:
Arduino programmed and operationally tested, tx-rx dmr mode is ok. I can
not test mode pins now because the system is remotely.

phl0, 2016-11-06:
Okay. I guess they are rarely used anyway. So I think we can close this issue then.

ik1whn, 2016-11-06:
ok thanks

closed 2016-11-06 by ik1whn