g4klx/MMDVM #43: Teensy 3.6 Support

phl0, 2016-11-28:
I successfully compiled the MMDVM firmware for a Teensy 3.6 board. However starting MMDVMCal exits with exit code 1 and the MMDVM obviously stops running due to the LED on pin 13 not flashing any more.

[image]

[image]

phl0, 2016-11-28:
Tracking the issue further in the MMDVM firmware. Commenting out line 290 "io.start() does not stop the MMDVM from running for now.

phl0, 2016-11-28:
If I comment out the "Initialise ADC0" part, the else part in the "#if defined(EXTERNAL_OSC)" section and the "Initialise DAC" part the MMDVM firmware seems to start. Using MMDVMCal I can then toggle the PTT at least once.

phl0, 2016-11-28:
@g4klx: Just a dirty guess... According to https://forum.pjrc.com/archive/index.php/t-24492.html it says that waiting for calibration is:

```
// Begin calibration
ADC0_SC3 = ADC_SC3_CAL;
// Wait for calibration
while (ADC0_SC3 & ADC_SC3_CAL);
```

So maybe that " != ADC_SC3_CAL" appedix is needles?

g4klx, 2016-11-29:
They are functionally the same. Mine is more explicit.

phl0, 2016-11-29:
Interesting because if I omit the "!= ADC_SC3_CAL" the firmware executes beyond this point. Otherwise it hangs while executing this part ...

g4klx, 2016-11-29:
Oh! I stand corrected.

Where does it stop now?

phl0, 2016-11-29:
Tested with latest commit a40506aff0c577819ef9a8ac03c0f531d7503ffc:

- MMDVMCal starts correctly
- I can toggle PTT on and off using space

phl0, 2016-11-29:
Successfully also tested the RX Part of the MMDVM on Teensy 3.6. Connected it to my XLX reflector and made some test QSOs.

phl0, 2016-11-29:
MMDVM on Teensy 3.2 also working. 

[image]

phl0, 2016-11-29:
Activating the external 12MHz TCXO in Config.h leads to MMDVMCal starting and beeing able to trigger the PTT once. There is no digital output on the TX NF line. After triggering the PTT once the MMDVM does not respond any more.

phl0, 2016-11-30:
Upgraded to 79440ca32df3ea8baa0f077dec3f458a9bf41479.
Compiled with 12 MHz TCXO enabled, connected LED to pin 3.
MMDVMCal can trigger the PTT only once, cannot de-key afterward. Pressing space multiple times gives:

[image]

No TX Audio while keyed and RX audio not decoding.

Disabling TCXO makes it work again.

N4IRS, 2016-12-12:
I connected TCXO to pin 13 of the Teensy 3.6. Got the same PTT issues. I'm seeing the on board LED light when TCXO connected. I wonder if the LED and current limiting resistor is causing a problem. I am considering removing the LED to continue testing.

g4klx, 2016-12-13:
I asked on the Teensy forum and was given some feedback on what was wrong. I've updated my code, so if you can try it again and let me know what happens. I think you might be right about the LED, it's a shame that you can't use another pin for it.

g4klx, 2016-12-14:
Things have definitely moved on, but not quite correct yet it seems. Using the external TCXO, the settings for things like the ADC and DAC levels should be exactly the same as for using the internal clock except it should be better for DMR use. The LED should flash at the same rate in both modes for example.

I've done a tweak to the code which I hope makes the external TCXO match the internal oscillator even better. Please keep testing and report back any differences between the two modes.

I wonder if there's any mileage in making pin 3 the LED in both modes and abandon the mounted LED?

g4klx, 2016-12-14:
That sounds like good news. I won't declare it working until someone has had a DMR QSO with the TCXO enabled! We also need to look at the RSSI side (also on the STM32) to make sure that it's working.

phl0, 2016-12-14:
Jepp sure. I thought I should get the RF side up and running before looking at the RSSI stuff.
Is the RSSI stuff ok for testing? I saw that you committed to that part of the code before.

g4klx, 2016-12-14:
Concentrate on the TCXO side. The RSSI *should* work as-is once you define the requirement for RSSI in the Config.h. I want to create a 48 kHz version as soon as possible for the ZUM v1.2 boards. I wonder if someone will do a Teensy shield?

g4klx, 2016-12-14:
I've done a small change based on further feedback, this may be even better :-)

N4IRS, 2016-12-14:
I am not having any luck on receive of any mode. No decode with or without the TCXO. The same ZUM board and Due work properly. Since others are able to receive, I can only assume I have something wrong.
I did verify receiver audio to Pin 14 (A0). I will check the level and double check my other connections.

phl0, 2016-12-14:
@N4IRS Pin 14 (A0) is correct for RX NF. Correct pin on the Teensy board? Is it a 3.6 one?

N4IRS, 2016-12-14:
Yes, I'm going to get my 3.6 working before I try the 3.2. I will re-check everything again tonight.

phl0, 2016-12-14:
And you are using the ZUM 1.0 board? I could cross check that if you like.

N4IRS, 2016-12-14:
Yes it is a ZUM 1.0 audio board. Transmit audio works, PTT works and a quick check with a audio amp proves receive audio on pin 14. Tonight I'll check the level at pin 14. The way I built my test bed, I can unplug the ZUM from the Due and plug it into a breadboard with the 3.6.

phl0, 2016-12-14:
Ok, I wired up the ZUM v1.0 board to the Teensy. Aligning now ...

phl0, 2016-12-14:
Huh, confirmed that the Teensy will not receive anything with the ZUM v1.0 board ... *confused*

N4IRS, 2016-12-14:
OK, That is just strange. Filter values on the ZUM board? I did try the wide filter setting in config.h

phl0, 2016-12-14:
Dunno. Re-plugged v1.2 board and it receives .....

N4IRS, 2016-12-14:
Do you have a schematic of the 1.2 board so we can compare to the 1.0?

N4IRS, 2016-12-14:
Do we have confirmation of a 1.0 board working with the STM?

g4klx, 2016-12-14:
I'd love to know the difference. AFAIK they have wider filters on the ZUM 1.2 which is why I'm going to do a modern 48 kHz version so that the Nyquist frequency is 24 kHz instead of 12 kHz.

N4IRS, 2016-12-14:
I THINK the 1.2 board filters are based on the EB4FBZ filter design.

phl0, 2016-12-14:
Thats right.

N4IRS, 2016-12-14:
I know Colin was selling some boards in the UK, but he did not want to ship to the US. I would love to buy some of the "other" boards out there so I can test. So far I have been unsuccessful getting anyone to sell me a board.

juribeparada, 2016-12-14:
Hi guys, I would like to help, but I still waiting for my Teensy 3.6 arriving. Regarding STM, all my tests have been done with MMDVM 1.0 board, working OK. At the moment I don't have the 1.2 board.

g4klx, 2016-12-14:
They gut who helped me with the LPTMR posted the following. Anybody willing to try it out?

[GitHub](https://github.com/manitou48/teensy3/blob/master/adclptmr.ino)

I clocked pin13 from PWM pin 23 for testing.

FWIW, this may be circular nonsense, but with my sketch (PWM/23 to pin 13) @48khz (CMR 1), the long term LPTMR count shows an error of 24 ppm. I have measured the crystal on this T3.2 at about -19 ppm (with NTP or GPS), so that might? suggest the sketch/LPTMR is adding an error of 5 ppm? (48khz PWM is exact prescaler of 96mhz, so no discretization error). if i feed 48khz PWM into FreqCount library, it counts exactly 48000 for every 1s sample period... probably meaningless.

I obviously don't have a signal generator. If you feed your signal generator into my sketch, what does sketch report as measured frequency? and if you feed your generator into a FreqCount library sketch? 

or not.

N4IRS, 2016-12-14:
I'll try it tonight and report back.

phl0, 2016-12-14:
So what is the task? Testing the sketch above with the 12MHz TCXO of a MMDVM boards? That yields:

[image]

phl0, 2016-12-14:
Also answered directly in the Teensy forum ;-)

phl0, 2016-12-15:
Tried manitou's sketch. Reports 17ppm ... Wierd .... Will try with the Teensy 3.2 and/or the ZUM 1.0 board

phl0, 2016-12-15:
Results are documented in the Teensy forum thread. Seems like we are measuring the drift of the Teensy's own crystal. Not sure if and how that affects the MMDVM's internals.

g4eml, 2016-12-19:
I have just got myself a Teensy 3.6. So far I am having the same problem where it doesn't appear to receive anything. The signal to A0 pin 14 looks fine on a scope and the level is well within the allowable range. I haven't connected up the TCXO yet and I am testing on both Dstar and DMR , neither shows any signs of receiving anything. Transmit is working fine so that indicates the interrupts must be firing at the correct rate (or close enough). I will do some more debugging to try to see what is happening. I can feel a lot of studying of the processor PDF document is going to be needed!

...Colin G4EML

g4klx, 2016-12-20:
That's interesting. I thought that Flo had it working fine on TX and RX a couple of weeks ago using the internal oscillator in D-Star mode. You are right about the interrupts though, the DAC is run off the ADC complete interrupt, which is hardware triggered off the timer. I also have a query about the RSSI ADC running, which is run from a software trigger on the signal ADC complete interrupt. but that can wait until later.

I get the feeling that it's almost there, but not quite. BTW I have pushed a small change to the calibration function based on feedback on the Teensy support forum which may help in some small way.

phl0, 2016-12-20:
Jonathan is right. At least with the internal oscillator (TCXO off) it
should work in D-Star more or less right away. I had that running a
while ago.

g4klx, 2016-12-20:
The Teensy RSSI input pin has changed. It turns out that ADC1 has only limited access to pins, and on the 3.1/3.2 they're only on the bottom side :-( The new mapping is: Teensy 3.5/3.6, A16, Pin 35. Teensy 3.1/3.2, A17, Pin 28. Please pass this information on to those that are testing it.

g4eml, 2016-12-20:
I modified teensyio.c so that the ADC values were immediately copied back out to the DAC and watched them on a scope. 
With noise on the Rx line the DAC signals were all max value (3.3V) with occasional spikes down to 1V.
I then fed in a variable DC voltage to the ADC and confirmed that the it appears to be working but with a reference voltage of about 1.2V instead of 3.3V. 
Any voltage above 1.2V  gives the 4095, the maximum reading.
As the centre point of the received audio is 1.65V then that would explain why my teensy is not receiving. 

I am now going back to the datasheet to try to see why this is happening. Maybe the internal reference voltage is programmable and has not been set up. 

Colin. 

From: Florian (DF2ET) 
Sent: Tuesday, December 20, 2016 8:50 AM
To: g4klx/MMDVM 
Cc: g4eml ; Comment 
Subject: Re: [g4klx/MMDVM] Teensy 3.6 Support (#43)

Jonathan is right. At least with the internal oscillator (TCXO off) it
should work in D-Star more or less right away. I had that running a
while ago.

g4eml, 2016-12-20:
The reference for both ADC0 and ADC1 was set to internal. The internal reference for the chip is 1.2V as suspected.  On the Teensy the external reference pins are connected to 3V3 so the reference should be set to external.  I changed this in and now Dstar and DMR are both working on internal clock. 
Next to have a look at TCXO. 
Pull request submitted.

g4eml, 2016-12-20:
Maybe that's also the reason the ZUM Boards were showing different behaviors. If one had the DC offset of the output below 1.2V it would probably work.

AllStarLink, 2016-12-20:
I'll try it tonight.

N4IRS, 2016-12-20:
Let's try this again. ;)

I'll try it tonight.

N4IRS, 2016-12-21:
Did some very quick tests. D-Star receives, Fusion receives, P25 receives. I think DMR will receive once I re-set TX levels. This is with a Teensy 3.6 and a ZUM 1.0 board.

N4IRS, 2016-12-21:
Tested with 14 MHz TCXO. No transmit. Onboard LED stays "lit" Did not test receive. I hesitate to remove the LED but I may have to. Still, a LOT of progress tonight.

g4klx, 2016-12-21:
Thank you for your great input Colin. I've merged the changes, as well as porting them to the 48 kHz branch. I do have one query, there are warnings in the documentation about using the same reference for the DAC and ADC(s) at the same time with issues of crosstalk. Is that going to be an issue here?

g4eml, 2016-12-21:
Hi Jonathan, 

I don't think it will be an issue. Most processor documents I have seen before include a similar warning, generally they are only talking about very small interactions affecting the least significant bits. 

I will look at the external TCXO next to see if I can get that working.

I am also looking forward to trying the 48Khz branch again. I think with the extra processing power available it will be worth pursuing to improve the sampling point resolution.

From: Jonathan Naylor 
Sent: Wednesday, December 21, 2016 8:58 AM
To: g4klx/MMDVM 
Cc: g4eml ; Comment 
Subject: Re: [g4klx/MMDVM] Teensy 3.6 Support (#43)

Thank you for your great input Colin. I've merged the changes, as well as porting them to the 48 kHz branch. I do have one query, there are warnings in the documentation about using the same reference for the DAC and ADC(s) at the same time with issues of crosstalk. Is that going to be an issue here?

g4klx, 2016-12-21:
The 48 kHz branch hasn't been tested and was put together quickly, so be careful with it. The levels are probably wrong at least :-(

g4klx, 2016-12-22:
Something work looking into in the future: getting ADC1 for RSSI to be run off the same hardware  interrupt as ADC0 to reduce the number of interrupts. This may give us a little more performance. It'd need to be done for both the PDB and LPTMR, but would bring the Teensy into line with the STM32 and the Due.

g4eml, 2016-12-22:
Tested with 14.4 Mhz TCXO. No ADC interrupts so no TX or Rx. 
Changed setting for ADC Pre-Trigger from B to A. (documentation is poor for this but I think A is correct for the way we are using it)
External TCXO now working. A 3 minute DMR transmission show no drift in the sample position as is correct. 
Pull Request submitted.

g4eml, 2016-12-22:
There is no need to make ADC1 use the PDB or LPTMR. 
Configure ADC1 as before but without enabling its interrupt. 
Every time ADC0 interrupts we read it and also read the previous conversion from ADC1. 
We then software trigger a new ADC1 conversion which will be available the next time ADC0 interrupts. 

I have tried this and it works fine as far as I can tell. 

Pull request created.

N4IRS, 2016-12-23:
Works well with the TCXO Transmit and receive on DMR and P25. Teensy 3.6.

g4klx, 2016-12-23:
The 48 kHz branch version of the Teensy is now out of date, could you do a similar pull request on that one too please.

g4klx, 2016-12-24:
I've updated the Teensy support on the 48 kHz branch to match the master branch apart from the sample rate.

[5 more replies]

closed 2017-02-14 by g4klx