g4klx/MMDVM #55: Compling error

dg3yjb, 2016-12-30:
During compiling with the latest Arduino IDE 1.8.0 I get the following errors:
Arduino: 1.8.0 (Linux), Board: "Arduino Due (Native USB Port)"

sketch/DMRDMOTX.cpp.o: In function `CDMRDMOTX::writeByte(unsigned char)':
sketch/DMRDMOTX.cpp:156: undefined reference to `arm_fir_fast_q15'
sketch/DMRTX.cpp.o: In function `CDMRTX::writeByte(unsigned char, unsigned char)':
sketch/DMRTX.cpp:299: undefined reference to `arm_fir_fast_q15'
sketch/DStarTX.cpp.o: In function `CDStarTX::writeByte(unsigned char)':
sketch/DStarTX.cpp:453: undefined reference to `arm_fir_fast_q15'
sketch/IO.cpp.o: In function `CIO::process()':
sketch/IO.cpp:185: undefined reference to `arm_fir_fast_q15'
sketch/IO.cpp:192: undefined reference to `arm_fir_fast_q15'
sketch/IO.cpp.o:sketch/IO.cpp:210: more undefined references to `arm_fir_fast_q15' follow
collect2: error: ld returned 1 exit status
exit status 1

g4klx, 2016-12-30:
Look at BUILD.txt contains details of other versions. It's probably correct for that one too.

dg3yjb, 2016-12-30:
With the update of the IDE there was an update of the Arduino SAM Board package to version 1.6.10. With this update the platform.txt was overwriten, with the changes discriped in the build.txt the compiling was succesfully.
Thanks for the support John.

closed 2016-12-30 by dg3yjb