g4klx/MMDVM #58: YSF RSSI Issue on Arduino DUE?

m1geo, 2017-01-09:
Hi All.

Seems to be an error with RSSI on YSF?

> M: 2017-01-09 20:44:00.991 YSF, received RF header from M5BXB      to ALL       
> M: 2017-01-09 20:44:01.583 YSF, transmission lost, 0.6 seconds, BER: 0.0%, RSSI: -150/-20/-41 dBm
> M: 2017-01-09 20:44:14.271 YSF, received RF header from M5BXB      to ALL       
> M: 2017-01-09 20:44:14.960 YSF, transmission lost, 0.7 seconds, BER: 0.0%, RSSI: -20/-20/-20 dBm
> M: 2017-01-09 20:44:16.481 YSF, received network data from GB7KH-RPT  to  ALL       at GB7KH     
> M: 2017-01-09 20:44:16.498 YSF, received network end of transmission, 0.7 seconds, 0% packet loss, BER: 0.0%

I know these users are not close enough to get -20 dBm reports.

g4klx, 2017-01-10:
The RSSI on D-Star, P25 and YSF operates on spot levels of the RSSI rather than an average like on DMR. I will make the changes to allow for averaging on those modes when I get a chance to do another update. Hopefully that will give more meaningful RSSI values.

m1geo, 2017-01-10:
Thanks Jon. George. M1GEO.

closed 2017-02-14 by g4klx