g4klx/MMDVM #64: Annoying problem !

iw9grl, 2017-02-14:
Hi dear, this is a question posted in Yahoo group :

One remaining issue affecting MMDVM in repeater operation is a "dead window" when the repeater should wake-up from "listening" status IMMEDIATELY after the ending of the TX hang time. This has been noticed in DMR mode on  the two totally different repeaters we built  for our ham club. 
This dead window is very narrow (perhaps ~200ms) but can randomly disrupt communication because the repeater doesn't wake up, and since the MS ( mobile station ) is still receiving the end of hang-time transmission, it assumes the repeater has waken-up so it keeps transmitting. When this happens, the whole voice message is lost.

Any solutions ?
Tnx 
73

g4klx, 2017-02-14:
The problem of the MMDVM not responding to DMR transmissions immediately after the transmitter stops is known about. The answer is not simple, in MMDVM terms.

As people have already identified, the user radio thinks that the loss of signal is because of local interference or terrain and assumes that the repeater is still operational and behaves accordingly. It doesn’t send the normal wakeup CSBK, until a while later when it decides that the repeater really has closed down. This behaviour while understandable, isn’t mandated in the standard which is why I didn’t put it into the MMDVM. It didn’t help that I developed the software in a vacuum without access to a real DMR repeater to compare with. Thankfully the DMR specification is very complete (do you hear me Icom and Yaesu).

With the MMDVM, some major changes would need to be made to enable the same sort of operation. As written the transmitter switches on as soon as data is ready to be transmitted, with DMR not only is data generated that is transmitted, but also timing information which isn’t transmitted but sent back to the receiver to enable TDMA operation. Splitting the operation of the transmitter from the presence of data wouldn’t be too complex, and would be an MMDVM firmware-only issue. This would allow for the timing information to be sent back to the receiver to allow it to operate. You’d probably need to send the data to the DMR idle receiver also for safety. This state of affairs would only last for a few hundreds of milliseconds before switching to true Idle mode.

The host would also need to be modified to handle normal DMR data as well as Idle data when in Idle mode. This would be relatively easy too.

It would be an interesting project for someone to do!

I would do it, but I have limited free time and I have other priorities with the MMDVM, i.e. the YSF and P25 demodulators.

iw9grl, 2017-02-14:
Thank you Jonathan for the detailed answer, I'm sorry but I don't have the skill to help you with the code!
I hope that someone read this message and try to find a solution. I'm always available to make any test if needed .

have a nice day
Daniele IW9GRL

ok2it, 2017-02-15:
Danielle does this happens if MMDVM is runing in DMR Only mode or in mixed DV Mode?

iw9grl, 2017-02-23:
It is not relevant, it happens only on dmr in the same way if it is in multimode or not

vk4tux, 2017-02-23:
Use TXHang=10 and Modehang-10 and you wont have the issue.

iw9grl, 2017-02-23:
Hi

I know this workaround to avoid this issue, also improve the Late Entry feature, but I think it is always a problem that need to be solved ( when it is possible :)  )  .

thank you
73 de IW9GRL

ve2nbz, 2017-02-23:
Interesting !! I'm test it now ! I have also the same behavior ! But i think this problem have to be solved in future update if possible !

vk4tux, 2017-02-23:
It's not a workaround, Its the best way to run your repeater. Keep TX condition between overs so the system is in TX for the whole qso. 100% duty cycle spec required from TX radio is the nature of DMR and provides best performance considering a 2nd qso/slot may also be happening. stop/start TX between overs has always been bad and we discussed / resolved this a year ago.

DMR is like a rollercoaster ride you jump in and out on the move, on the slot you like

iw9grl, 2017-02-23:
Thank you for the detailed information. 73

vk4tux, 2017-03-02:
The issue really began for many with the separate RFHang and NetHang parameters added to replace ModeHang. The new parameters reduced TXhang less than its setting to the lowest common denominator. 

Using original Modehang and organising best mode setups to not take over. (dstar reflector link auto timed disconnect on cron schedule) etc make the system function much better for all using 10 seconds for Mode/TX hang.

10 seconds is the TX/Mode Hang  time setting in practise we have found, running 4 mode systems.

ok2it, 2017-03-02:
That is very interesting,
Not that it resolves the behavior, but as a Best-Practice guide, I guess this might be useful to add to BM WiKi or MMDVM WiKi.
I have OK0DIT RPTR with D* with ircddbgw,DMR BM and YSF with YSFGW.
I have the TX Hang at 5 and Net hang at 15 and RF Hang at 30.
That way all works well – but I am tempted to put 10s in the mode hang and and REM out RF/NETmodes.

YSFGW is disconnected,  ircddbgw is linked as RPTR is used as point of listening to on D* and DMR.

I will be honest but I haven’t yet ran into issue described here – I had to focus on Txing right after TXTail  has dropped off so I don’t consider it to be much of an issue over here.

Jiri

vk4tux, 2017-03-02:
The behavior is only seen when the repeater is used with wrong 
configuration.

The sharp stick at the bottom of the stream is no issue for those 
floating across and hence is of no consequence.

closed 2017-03-13 by g4klx