g4klx/MMDVM #91: MMDVM/BUILD.txt - Arduino 1.8.2

mrickey, 2017-05-04:
I'm trying to build the code that runs on the Due.

In the Build.txt file I see instructions for editing the platform.txt file. This is fine.

In version 1.8.2, this file seems to have moved. It is not in the AppData folder tree any longer. I found it in the install folder C:\Program Files (x86)\Arduino\. In fact, there are two files with this name and they have different content.

C:\Program Files (x86)\Arduino\hardware\platform.txt (1kb)
C:\Program Files (x86)\Arduino\hardware\arduino\avr\platform.txt

I don't understand the reference to SAM in BUILD.txt...is this something I need to add to my Arduino IDE install?

In the instructions I'm asked to add the math library to a section in platform.txt. I found the section, but the line to be modified is much shorter...is there a different modification that needs to be done?

And finally, after making the change I tried to Verify the project and I get the following errors:

Arduino: 1.8.2 (Windows 10), Board: "Arduino/Genuino Uno"

In file included from C:\Users\Michael\Google Drive\Projects\g4klx\MMDVM\MMDVM.ino:21:0:

> Globals.h:36: error: #error "Unknown processor type"
>  #error "Unknown processor type"
>   ^
> sketch\Globals.h:39:22: fatal error: arm_math.h: No such file or directory
>  #include <arm_math.h>
>                       ^
> compilation terminated.
> exit status 1
> #error "Unknown processor type"
> 

I'm not sure how to proceed...suggestions?

phl0, 2017-05-04:
The two platform.txt that you mentioned are definitely the wrong ones. The right one is simply not there because you didn't install the hardware (CPU) support for the Arduino Due boards in the IDE. 
In the board manager you have to select and install the "Arduino SAM Boards (32-bits ARM Cortex-M3). That should allow you to select the Arduino Due under boards. And you should have the platform.txt in the App Data folder.

mivsek, 2017-05-10:
I just compiled successfully on Arduino 1.8.2 by editing platform.txt in:
c:\Users\Janko Mivšek\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11

Bob-VK6ZGN, 2017-05-17:
Hi Guys Iam getting the following error on Arduino 1.8.2
arm-none-eabi-gcc: error: C:\Users\Bob\AppData\Local\Temp\arduino_build_508069/syscalls_sam3.c.o: No such file or directory

exit status 1
Error compiling for board Arduino Due (Programming Port).
Any thoughts ??
Bob

phl0, 2017-05-17:
Never seen that before. Did you install support for the SAM boards (see above)?

Bob-VK6ZGN, 2017-05-17:
Hi
Yes I installed the info for the Duo. and can select it in the boards? list 
 both the normal and the program ports

So not sure what's going on ?

Bob

mrickey, 2017-05-22:
@Bob-VK6ZGN You said you installed the info for the "Duo" is that a typo? You should have installed support for the "Due". These boards have different processors.

@phl0 Thank you for your help. I was able to find the correct platform.txt and I have since loaded the firmware.

closed 2017-05-22 by g4klx

Bob-VK6ZGN, 2017-05-23:
Hi Michael

Yes It must have been a typo I did install the DUE

Still issues ! On that pc

However I downloaded a complete new installation of the IDE on a second 
laptop and modified the file..   Different errors this time ?  So I looked 
very carefully at the mods to the line in the file and found the order of 
the modification was different to one of the samples
So I corrected it   Now seems to compile without errors.  !!!  On to the 
next challenge

Tks for your help

Bob

afa7mv, 2018-04-08:
I am trying to use a ardiuno uno with pi and mmdvm board having trouble with firmware.

juribeparada, 2018-04-08:
Hi afa7mv: Arduino Uno is not supported for MMDVM, just Arduino Due.

ra1amx, 2019-03-20:
Hello. Some troubles detectid in log. Can help me ? 
[error log.txt](https://github.com/g4klx/MMDVM/files/2989743/error.log.txt)

ra1amx, 2019-03-20:
Accept - I work on latest de, but need 1.6.9 - then all ok install on DUE
[image]