g4klx/MMDVM #95: GitVersion.h not generated inside Arduino Makefile

m1geo, 2017-06-03:
I use `make -f Makefile.Arduino upload` inside the MMDVM root folder.

It seems that GitVersion.h 

> SerialPort.cpp:24:24: fatal error: GitVersion.h: No such file or directory
> #include "GitVersion.h"
>                        ^
> compilation terminated.
> Makefile.Arduino:113: recipe for target '/home/george/Arduino/MMDVM/build/SerialPort.cpp.o' failed
> make: *** [/home/george/Arduino/MMDVM/build/SerialPort.cpp.o] Error 1
> george@watt:~/Arduino/MMDVM$ make -f Makefile.Arduino upload

m1geo, 2017-06-03:
Aaah, it seems you have to run compile first, before upload. Perhaps we need to add GitVersion.h to the dependency for upload, since it builds with the upload.

phl0, 2017-06-03:
Hhmmm. I think it does not make sense to make GitVersion.h a dependency of upload because that should be compiled into the binary and upload does only handle the binary. GitVersion.h should only be needed for compile IMHO.

But I have no clue why upload fails when GitVersion.h is not existent ... Will investigate...

m1geo, 2017-06-04:
I disagree. Issuing `make upload` should do everything up to and including the upload. A _compile_ should be a requirement of an _upload_.

I worked around it, it's not  a huge issue. It just may confuse some.

I was on the repeater site and didn't have time to look yesterday. Thanks @phl0 :+1:

phl0, 2017-06-06:
https://github.com/g4klx/MMDVM/pull/97 should fix it.
@m1geo pls cross-check.

m1geo, 2017-06-06:
#97 Fixes issue. Thanks @phl0

closed 2017-06-06 by m1geo