g4klx/MMDVM #101: Audio loss on MotoTRBO XPR series

va2na, 2017-07-31:
Hi Jonathan,

We have a couple of user that have audio loss problem on motorola XPR 6550-4500 series. 

Audio loss is just after receiving ID from the repeater. and held 2 secondes,  Talker Alias is disabled on each repeater and we still got the issue.

Can send video for more detail.

Thank you

dl7ata, 2017-08-25:
I got the same effect with the similar DP3600 on my hotspot.
Audio is loss for about 1 sec. and I can't receive any SMS (caused by this delay?).
Regards
Frank

closed 2018-02-02 by va2na

citynetcomputing, 2018-04-15:
Same issue here... XPR7550e.

dl7ata, 2018-04-15:
Whats your jitter-value in your MMDVM.conf ?

citynetcomputing, 2018-04-15:
standard 300

citynetcomputing, 2018-04-15:
its RF to RF anyway. Has nothing to do with Jitter. ;)

dl7ata, 2018-04-15:
What do you mean by "RF to RF"? 
Set it to zero, this should work for the audio delay but not for missing SMS.

citynetcomputing, 2018-04-15:
Radio to Radio through the Repeater. No internet, No linking. etc..

dl7ata, 2018-04-15:
I don't know your setup. How is the digital stream running? From one radio strait into another? In that case forget jitter.

citynetcomputing, 2018-04-15:
No Its an MMDVM PI based DMR repeater.

dl7ata, 2018-04-15:
Did you try jitter=0 ?

citynetcomputing, 2018-04-15:
Jitter is only for the DMR gateway link so no. Has nothing to do with my issue but thanks!