g4klx/MMDVM #105: Got a NAK from the modem

SkyAndy, 2017-08-20:
the last 2 merge does not work on my Teensy 3.6 on MMDVM Teensy V1.1 Bord.
only to 8f317fd332992e410a399af47d6244f933edce42 merge
Hardware:
Teensy 3.6 / 180Mhz / Optimize Faster
compiled on:
MAC El Capitan with Arduino 1.6.12 (OSX) with Teensyduino 1.36

juribeparada, 2017-08-21:
I tested with a Teensy 3.5 and works good. Could you copy your NAK message?. If you received: "Received a NAK to the SET_CONFIG command from the modem", you need to update your MMDVMHost also. Jonathan added a new parameter to the modem configuration.

phl0, 2017-08-21:
No problems here with latest software versions:

- MMDVMHost: commit 0efaf55102489c0f58e2e0dffa94b25cf8bbc646
- MMDVM: commit 5b1075f0f789ab0297561783fddf237f4764bfb2
- Arduino: 1.8.3
- Teensyloader 1.37

''
$ ./MMDVMHost MMDVM.ini 
I: 2017-08-21 07:07:01.527 This software is for use on amateur radio networks only,
I: 2017-08-21 07:07:01.527 it is to be used for educational purposes only. Its use on
I: 2017-08-21 07:07:01.527 commercial networks is strictly prohibited.
I: 2017-08-21 07:07:01.527 Copyright(C) 2015-2017 by Jonathan Naylor, G4KLX and others
M: 2017-08-21 07:07:01.527 MMDVMHost-20170719 is starting
M: 2017-08-21 07:07:01.527 Built 09:04:41 Aug 21 2017 (GitID #0efaf55)
I: 2017-08-21 07:07:01.527 General Parameters
I: 2017-08-21 07:07:01.527     Callsign: G9BF
I: 2017-08-21 07:07:01.527     Id: 123456
I: 2017-08-21 07:07:01.527     Duplex: yes
I: 2017-08-21 07:07:01.527     Timeout: 180s
I: 2017-08-21 07:07:01.527     RF Mode Hang: 10s
I: 2017-08-21 07:07:01.527     Net Mode Hang: 3s
I: 2017-08-21 07:07:01.527     D-Star: enabled
I: 2017-08-21 07:07:01.527     DMR: enabled
I: 2017-08-21 07:07:01.527     YSF: enabled
I: 2017-08-21 07:07:01.527     P25: enabled
I: 2017-08-21 07:07:01.527 Modem Parameters
I: 2017-08-21 07:07:01.527     Port: /dev/ttyACM3
I: 2017-08-21 07:07:01.527     RX Invert: no
I: 2017-08-21 07:07:01.527     TX Invert: yes
I: 2017-08-21 07:07:01.527     PTT Invert: no
I: 2017-08-21 07:07:01.527     TX Delay: 100ms
I: 2017-08-21 07:07:01.527     RX Offset: 0Hz
I: 2017-08-21 07:07:01.527     TX Offset: 0Hz
I: 2017-08-21 07:07:01.527     TX DC Offset: 0
I: 2017-08-21 07:07:01.527     DMR Delay: 0 (0.0ms)
I: 2017-08-21 07:07:01.528     RX Level: 50.0%
I: 2017-08-21 07:07:01.528     CW Id TX Level: 50.0%
I: 2017-08-21 07:07:01.528     D-Star TX Level: 50.0%
I: 2017-08-21 07:07:01.528     DMR TX Level: 50.0%
I: 2017-08-21 07:07:01.528     YSF TX Level: 50.0%
I: 2017-08-21 07:07:01.528     P25 TX Level: 50.0%
I: 2017-08-21 07:07:01.528     RX Frequency: 435000000Hz (435000000Hz)
I: 2017-08-21 07:07:01.528     TX Frequency: 435000000Hz (435000000Hz)
M: 2017-08-21 07:07:01.528 Opening the MMDVM
I: 2017-08-21 07:07:03.543 MMDVM protocol version: 1, description: MMDVM 20170501 TCXO (D-Star/DMR/System Fusion/P25/RSSI/CW Id) (Build: 09:03:07 Aug 21 2017)
I: 2017-08-21 07:07:03.563 Display Parameters
I: 2017-08-21 07:07:03.563     Type: None
I: 2017-08-21 07:07:03.563 D-Star Network Parameters
I: 2017-08-21 07:07:03.563     Gateway Address: 127.0.0.1
I: 2017-08-21 07:07:03.563     Gateway Port: 20010
I: 2017-08-21 07:07:03.563     Local Port: 20011
M: 2017-08-21 07:07:03.563 Opening D-Star network connection
I: 2017-08-21 07:07:03.563 DMR Network Parameters
I: 2017-08-21 07:07:03.563     Address: 44.131.4.1
I: 2017-08-21 07:07:03.563     Port: 62031
I: 2017-08-21 07:07:03.563     Local: random
I: 2017-08-21 07:07:03.563     Jitter: 300ms
I: 2017-08-21 07:07:03.563     Slot 1: enabled
I: 2017-08-21 07:07:03.563     Slot 2: enabled
I: 2017-08-21 07:07:03.563 Info Parameters
I: 2017-08-21 07:07:03.563     Callsign: G9BF
I: 2017-08-21 07:07:03.563     RX Frequency: 435000000Hz
I: 2017-08-21 07:07:03.563     TX Frequency: 435000000Hz
I: 2017-08-21 07:07:03.563     Power: 1W
I: 2017-08-21 07:07:03.563     Latitude: 0.000000deg N
I: 2017-08-21 07:07:03.563     Longitude: 0.000000deg E
I: 2017-08-21 07:07:03.563     Height: 0m
I: 2017-08-21 07:07:03.563     Location: "Nowhere"
I: 2017-08-21 07:07:03.563     Description: "Multi-Mode Repeater"
I: 2017-08-21 07:07:03.563     URL: "www.google.co.uk"
M: 2017-08-21 07:07:03.563 DMR, Opening DMR Network
I: 2017-08-21 07:07:03.564 System Fusion Network Parameters
I: 2017-08-21 07:07:03.564     Local Address: 127.0.0.1
I: 2017-08-21 07:07:03.564     Local Port: 3200
I: 2017-08-21 07:07:03.564     Gateway Address: 127.0.0.1
I: 2017-08-21 07:07:03.564     Gateway Port: 4200
M: 2017-08-21 07:07:03.564 Opening YSF network connection
I: 2017-08-21 07:07:03.564 P25 Network Parameters
I: 2017-08-21 07:07:03.564     Gateway Address: 127.0.0.1
I: 2017-08-21 07:07:03.564     Gateway Port: 42020
I: 2017-08-21 07:07:03.564     Local Port: 32010
M: 2017-08-21 07:07:03.564 Opening P25 network connection
I: 2017-08-21 07:07:03.564 CW Id Parameters
I: 2017-08-21 07:07:03.564     Time: 10 mins
I: 2017-08-21 07:07:03.564     Callsign: G9BF
I: 2017-08-21 07:07:03.564 RSSI
I: 2017-08-21 07:07:03.564     Mapping File: RSSI.dat
I: 2017-08-21 07:07:03.564 Loaded 0 RSSI data mapping points from RSSI.dat
I: 2017-08-21 07:07:03.564 DMR Id Lookups
I: 2017-08-21 07:07:03.564     File: DMRIds.dat
I: 2017-08-21 07:07:03.564     Reload: 24 hours
I: 2017-08-21 07:07:03.616 Loaded 67992 Ids to the callsign lookup table
I: 2017-08-21 07:07:03.616 D-Star Parameters
I: 2017-08-21 07:07:03.616 Started the DMR Id lookup reload thread
I: 2017-08-21 07:07:03.616     Module: C
I: 2017-08-21 07:07:03.616     Self Only: no
I: 2017-08-21 07:07:03.616     Ack Reply: yes
I: 2017-08-21 07:07:03.616     Ack Time: 750ms
I: 2017-08-21 07:07:03.616     Error Reply: yes
I: 2017-08-21 07:07:03.616 DMR Parameters
I: 2017-08-21 07:07:03.616     Id: 123456
I: 2017-08-21 07:07:03.617     Color Code: 1
I: 2017-08-21 07:07:03.617     Self Only: no
I: 2017-08-21 07:07:03.617     Embedded LC Only: no
I: 2017-08-21 07:07:03.617     Dump Talker Alias Data: yes
I: 2017-08-21 07:07:03.617     Prefixes: 0
I: 2017-08-21 07:07:03.617     Call Hang: 3s
I: 2017-08-21 07:07:03.617     TX Hang: 3s
I: 2017-08-21 07:07:03.617 YSF Parameters
I: 2017-08-21 07:07:03.617     Low Deviation: no
I: 2017-08-21 07:07:03.617     Remote Gateway: no
I: 2017-08-21 07:07:03.617     Self Only: no
I: 2017-08-21 07:07:03.618     DSQ: no
I: 2017-08-21 07:07:03.618 P25 Parameters
I: 2017-08-21 07:07:03.618     Id: 123456
I: 2017-08-21 07:07:03.618     NAC: $293
I: 2017-08-21 07:07:03.618     UID Override: no
I: 2017-08-21 07:07:03.618     Self Only: no
M: 2017-08-21 07:07:03.618 MMDVMHost-20170719 is running
''

dg9vh, 2017-08-21:
NAK-Messages could be reproduced with lastest MMDVMCal-Tool. Here also tested and NAK from MMDVMCal but MMDVMHost running fine best without any messages.

SkyAndy, 2017-08-21:
works fine after change ini. only MMDVCal tool work not.

closed 2017-08-21 by SkyAndy

g4klx, 2017-08-21:
Fixed MMDVMCal. Thanks for reporting it.