g4klx/MMDVM #127: DMR Rx Does Not Recognize Transmit Interrupt Window

aminhusni, 2017-11-25:
Affected function: 
DMR

Suspected problem description:
The firmware does not recognize that certain window of silent frame/window in a voice transmission is actually an intentionally implemented feature in DMR radios. 

[image]

As we can see here in the diagram, we could see that on every second, there is a 'missing' voice packet on the active slot being used. The function of this empty window is to allow for the transmitting radio to give a window to listen for if there is any other radio trying to transmit at the same frequency and slot. 

This feature is implemented so that the transmitting radio could de-key itself and allow for the interruption to occur. This is so that urgent traffic such as transmission coming from a supervisor's radio could override the current transmission on the same frequency. This feature is implemented in various radios such as Hytera and [Motorola models](http://www.radiocoms.co.uk/security/how-can-mototrbo-transmit-interrupt-help-your-communications). 

Replication of problem:
In my setup, I am using a Mototrbo radio specifically XiR 8668.
Enable "Allow Interruption" for the channel in your codeplug. 
Start transmitting to the hotspot.

Problem appearance: 
Voice degradation. 
Bit Error Rate > 1.1% - 2.0%
The normal bit error rate is 0.2% - 0.5% when the feature is turned off for the channel in the radio.

This made me suspect that the receiving part of the DMR code firmware is not properly recognizing this small window of empty voice frame. By treating it as an error, voice quality suffers at the Parrot, Hoseline and other receivers.

WillComHN, 2017-12-01:
Saludes a todos, estoy usando dos radios mototrbo dgp4100+,  sin privacidad funciona bien, pero cuando uso codigo de privacidad avanzado,  hay momentos donde se desengancha la comunicacion y eso es muy molesto, que se puede hacer?, att William Pineda desde Honduras C.A.

citynetcomputing, 2018-04-14:
I am having the same issue with MMDVM repeater, MD380 talking to a XPR7550e. XPR keeps dropping.