g4klx/MMDVM #131: Kenwood NX-5300 can not works with MMDVM Repeater in DMR Mode

hoky, 2017-12-25:
I have NX-5300 and Xir P8668/XPS 7550 (us area name)

P8668 works well with MMDVM, but NX-5300 got problems.

I have contacted with the Kenwood support and told me to check the Repeater.

here is the Manual about work with Repeater.

https://image.ibb.co/ny5yc6/QQ_20171226001000.png

g4klx, 2017-12-29:
I think something must be wrong with your system. I too have an NX-5300 and it works with full-duplex MMDVM repeaters, simplex MMDVM systems, and MMDVM hostspots including the ZUMSpot and MMDVM_HS.

hoky, 2017-12-29:
@g4klx 
Can you help me to see the problem? I posted this question in the radioreference.com kenwood forum  and asked the dealer, they do not know why. 

My XPS 7550 use "All Call" settings can be used normally.

I have been bothered by this problem for a week.

Screenshot of my configuration:
http://image.ibb.co/juBB2G/QQ_20171230012027.png  
http://image.ibb.co/i11N9w/QQ_20171230011949.png

hoky, 2017-12-30:
Check out the problem is the frequency of interference, Thank you.

closed 2018-09-18 by phl0