g4klx/MMDVM #132: High RX BER mmdvm

5b4amd, 2017-12-26:
Hi all,
i am using the following:

mmdvm zum
Yaesu 7800
rpi 3
arduino due 

I notice that i have high RX BER and i mean values between 0.5% to 3.2% (with fluctuations). Any ideas? Is the 7800 causing this? or do i need to check something (config) on the mmdvm board?

Thanks,
5B4AMD 73s

DL4OCH, 2017-12-27:
Yaesu 7800 doesn't perform well for DMR.

Best,
 DL4OCH

5b4amd, 2017-12-29:
Thanks for the information. Can you please let me know (if you are aware) which transceiver operates well with DMR. I 've seen around some motorola.

Thanks in advance,
5B4AMD 73s

phl0, 2017-12-29:
I tested Motorola GM3x0, GM1200E and a Tait TB7100 successfully. There is also a list on the Brandmeister Wiki.

5b4amd, 2018-02-02:
Hi all,
i am currently using for my repeater a FT-7800 for TX and a FT-1980 for RX (needs mod) with amazing results ,0% BER both ways.

Thanks all for your support,
73s de 5B4AMD
George

closed 2018-02-02 by 5b4amd