g4klx/MMDVM #135: How to enable debug message?

jcyfkimi, 2018-01-04:
I build the code for STM32F411RE-NUCLEO board with below configurations in config.h
#define EXTERNAL_OSC 12288000    //I'm using FOX924B-12.288 as tcxo
#define ARDUINO_MODE_PINS
#define STM32F4_NUCLEO_MORPHO_HEADER
#define SEND_RSSI_DATA
Unfortunately it's not working as expected after I upload the hex file to the board, the live led not blinking and the MMDVMHost application can't connect to the board, so I want to check if there's any to enable some debug messages printed on serial port? I tried to open the serial port but nothing shows.

juribeparada, 2018-01-04:
Your settings are correct, but STM32F411 is not supported by the current MMDVM firmware. The only STM32 parts supported are: STM32F407, STM32F446, STM32F722 and STM32F767.

jcyfkimi, 2018-01-04:
That's interesting, I mean both F411 and F446 are included in F4 families, so what's the limitation of 411 board? The only reason I can imaging is that there's some key features that MMDVM firmware used but the 411 chip don't have, what are they?

juribeparada, 2018-01-04:
The F411 doesn't have DAC, an important component for MMDVM.

jcyfkimi, 2018-01-05:
Noticed and thanks for you information.

closed 2018-10-10 by juribeparada