g4klx/MMDVM #142: Change time format for Nextion display

gm8arv, 2018-01-31:
I'm running an MMDVM_HS_HAT with a Nextion display. I want to produce a more UK-friendly clock on the display with the format:

2018-Jan-30 17:05:22

so that (a) the 2018-05-06 US day/month ambiguity is resolved and (b) the AM/PM time is replaced with the 24-hour clock. I have located the file and the lines to be changed (Nextion.cpp, line:729) and would like to have a go at this, but when downloading the Git and compiling I get a 3,183,132 byte MMDVMHost executable file whereas the working file in /usr/local/bin on my system is 3,299,580 bytes.  I edited the .cpp and ran make, but the executable doesn't run, in that on Pi-Star Modes Enabled shows DMR in red [the only mode I use] and there nothing changes on the display. Suspect I'm missing some make conditionals but it's not obvious what!

Can anyone help, please?

Also posted at: https://groups.io/g/MMDVM-HS-Hat/topic/compiling_mmdvmhost/9499033?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,9499033

gm8arv, 2018-02-01:
Someone (name lost, sorry) suggested making with:

make -f Makefile.Pi

That did produce a very slightly larger binary (3184420 bytes), but still doesn't work.  Same result as before (red DMR button).

g0wfv, 2018-02-02:
When I wrote the code for this, the date and time formats were made to follow the system locale, so is there is no ambiguity. Also, 12/24 time format is selectable; I can't remember quite good, but I remember testing it with DF2ET.

Bottom line, you may have something configured incorrectly on your system if you're running into date time format issues.

Make sure your system locale is set to en_GB not en_US.

````
╔═══════════════════════════════════════════════════════════════════════╗
║                                                                       ║
║           ██████╗ ██╗      ███████╗████████╗ █████╗ ██████╗           ║
║           ██╔══██╗██║      ██╔════╝╚══██╔══╝██╔══██╗██╔══██╗          ║
║           ██████╔╝██║█████╗███████╗   ██║   ███████║██████╔╝          ║
║           ██╔═══╝ ██║╚════╝╚════██║   ██║   ██╔══██║██╔══██╗          ║
║           ██║     ██║      ███████║   ██║   ██║  ██║██║  ██║          ║
║           ╚═╝     ╚═╝      ╚══════╝   ╚═╝   ╚═╝  ╚═╝╚═╝  ╚═╝          ║
║                                                                       ║
╚════════════════════════════════════════════════════════════════v3.4.8═╝
From your Windows Computer:
Pi-Star Dashboard:      http://pi-star/

From your Apple iPhone, iPad, Macbook, iMac etc.
Pi-Star Dashboard:      http://pi-star.local/

Pi-Star's disk is read-only by default, enable read-write with "rpi-rw".
Pi-Star built by Andy Taylor (MW0MWZ), pi-star tools all start "pistar-".

pi-star@pi-star(rw):~$ locale
LANG=en_GB.UTF-8
LANGUAGE=
LC_CTYPE="en_GB.UTF-8"
LC_NUMERIC="en_GB.UTF-8"
LC_TIME="en_GB.UTF-8"
LC_COLLATE="en_GB.UTF-8"
LC_MONETARY="en_GB.UTF-8"
LC_MESSAGES="en_GB.UTF-8"
LC_PAPER="en_GB.UTF-8"
LC_NAME="en_GB.UTF-8"
LC_ADDRESS="en_GB.UTF-8"
LC_TELEPHONE="en_GB.UTF-8"
LC_MEASUREMENT="en_GB.UTF-8"
LC_IDENTIFICATION="en_GB.UTF-8"
LC_ALL=
pi-star@pi-star(rw):~$
````

g0wfv, 2018-02-02:
When I wrote the code for this, the date and time formats were made to follow the system locale, so is there is no ambiguity. Also, 12/24 time format is selectable; I can't remember quite good, but I remember testing it with DF2ET.

Bottom line, you have something configured incorrectly on your system!

Make sure your system locale is set to en_GB not en_US.

````
╔═══════════════════════════════════════════════════════════════════════╗
║                                                                       ║
║           ██████╗ ██╗      ███████╗████████╗ █████╗ ██████╗           ║
║           ██╔══██╗██║      ██╔════╝╚══██╔══╝██╔══██╗██╔══██╗          ║
║           ██████╔╝██║█████╗███████╗   ██║   ███████║██████╔╝          ║
║           ██╔═══╝ ██║╚════╝╚════██║   ██║   ██╔══██║██╔══██╗          ║
║           ██║     ██║      ███████║   ██║   ██║  ██║██║  ██║          ║
║           ╚═╝     ╚═╝      ╚══════╝   ╚═╝   ╚═╝  ╚═╝╚═╝  ╚═╝          ║
║                                                                       ║
╚════════════════════════════════════════════════════════════════v3.4.8═╝
From your Windows Computer:
Pi-Star Dashboard:      http://pi-star/

From your Apple iPhone, iPad, Macbook, iMac etc.
Pi-Star Dashboard:      http://pi-star.local/

Pi-Star's disk is read-only by default, enable read-write with "rpi-rw".
Pi-Star built by Andy Taylor (MW0MWZ), pi-star tools all start "pistar-".

pi-star@pi-star(rw):~$ locale
LANG=en_GB.UTF-8
LANGUAGE=
LC_CTYPE="en_GB.UTF-8"
LC_NUMERIC="en_GB.UTF-8"
LC_TIME="en_GB.UTF-8"
LC_COLLATE="en_GB.UTF-8"
LC_MONETARY="en_GB.UTF-8"
LC_MESSAGES="en_GB.UTF-8"
LC_PAPER="en_GB.UTF-8"
LC_NAME="en_GB.UTF-8"
LC_ADDRESS="en_GB.UTF-8"
LC_TELEPHONE="en_GB.UTF-8"
LC_MEASUREMENT="en_GB.UTF-8"
LC_IDENTIFICATION="en_GB.UTF-8"
LC_ALL=
pi-star@pi-star(rw):~$
````

g0wfv, 2018-02-02:
Also I just realised ....

The date and time formats have nothing to do with the code in this repo (MMDVM); you really ought to have posted in [MMDVMHost.](https://github.com/g4klx/MMDVMHost/issues)

gm8arv, 2018-02-02:
Tony, many thanks for that.  I've never played with Linux locales, but after your suggestion I checked and found mine was US (this from the default Pi-Star image download), so I used raspi-config (which I needed to install), selected the US and GB locales to generate, selected the GB locale to run, rebooted and I'm getting UK times (can't check the date as 2/2 is correct on both sides of the pond!).  I'll try and repost in the correct repo (I thought I had but ...).  This raises another question:

Can I reset the system date format to be what I want in Linux, so your code doesn't need altering?

Background: I would like to make simple contributions but thought that as a starter I should at least be able to recompile the present code successfully, and then make one small change (just for myself) to prove I could make changes.

73, David GM8ARV

g0wfv, 2018-02-02:
Makefile.Pi includes the wiringPi library needed for the HD44780 and OLED LCDs - a red herring in your case I'm afraid.

The date and time formats are, I'm afraid, not changeable, other than through changing the locale.

We (@phl0 and I!) experimented extensively with different methods of changing the date and time formats, and linking to the system locale came out as the best compromise of portability and regional variation.

Sounds like yours is now sorted though :-)

gm8arv, 2018-02-02:
Makefile.Pi includes the wiringPi library needed for the HD44780 and OLED 
LCDs - a red herring in your case I'm afraid.

The date and time formats are, I'm afraid, not changeable, other than 
through changing the locale.

We (@phl0 and I!) experimented extensively with different methods of 
changing the date and time formats, and linking to the system locale came 
out as the best compromise of portability and regional variation.

Sounds like yours is now sorted though :-)
======================

Noted on the Makefile.Pi, Tony.

Yes, the date & time aspect is /almost/ resolved, although I still want to 
display the month.  A pity that Linux doesn't support the same system-wide 
settings as Windows.

However, I'm still stuck with being unable to compile a working version of 
MMDVMHost.  Until I can do that I can't make any (tested) contributions.

73,
David GM8ARV
-- 
SatSignal Software - Quality software written to your requirements
Web: http://www.satsignal.eu
Email: david-taylor@blueyonder.co.uk
Twitter: @gm8arv