g4klx/MMDVM #144: Audio loss on MotoTRBO XPR series

va2na, 2018-02-02:
Hello Jonathan,

Motorola user experience audio loss and DMR id loss of few sec at begin of Voice call intermittantly.  
maybe 10% ratio.

This happen from RF to RF communcation and i found that using TYT debug mode and DSD+ what happen each time.  The MMDVM send 3 voice header follow by the super voice frame, but when the glitch occur there 1, 2 or 3 idle frame added between the 3 Voice hdr, just look in the picture..and i confirm...each time we loss the audio in the Motorola...there an unwainted idle frame between the 3 Voice Hdr.

I hope it help

Guillaume
[image]

va2na, 2018-02-02:
Another one

[image]

aminhusni, 2018-02-07:
Have you checked Issue #127 ? Might be related.

citynetcomputing, 2018-04-14:
Same issue, MMDVM Repeater, TYT MD380 cannot talk to XPR7550e. XPR keeps dropping.

citynetcomputing, 2018-04-14:
anyone working on this issue??

closed 2018-10-08 by juribeparada