g4klx/MMDVM #154: pistar-mmdvmcal is not working

xe2pmp, 2018-02-23:
Hi guys!
I updated the mmdvm firmware 02/23/2018, and the pistar-mmdvmcal is not working.
Version: 1 "MMDVM 20180223 TCXO (D-Star/DMR/System Fusion/P25/NXDN/RSSI/CW Id) (Build: 15:34:57 Feb 23 2018)"
Got a NAK from the modem
Starting Pi-Star Services...
Finnished...

juribeparada, 2018-02-24:
You need to update your MMDVMCal too. Probably Pi-Star version of MMDVMCal is still old.

xe2pmp, 2018-02-24:
Hi Andy!
In fact, it stopped working after compiling and installing the new software for mmdvm.
Previosly just run the sudo pistar-mmdvmcal and that´s it.
Well, I have to find out how to do it.
regards
Pablo
xe2pmp
  ----- Original Message ----- 
  From: Andy CA6JAU 
  To: g4klx/MMDVM 
  Cc: xe2pmp ; Author 
  Sent: Friday, February 23, 2018 6:08 PM
  Subject: Re: [g4klx/MMDVM] pistar-mmdvmcal is not working (#154)

  You need to update your MMDVMCal too. Probably Pi-Star version of MMDVMCal is still old.

xe2pmp, 2018-02-24:
Done,
After of updated the MMDVMCal, it is working.

closed 2018-02-24 by xe2pmp