g4klx/MMDVM #158: RSSIRB.h:37:20: error: expected ')' before 'length'  until compile for STM32F105RBT MCU

badtime, 2018-03-03:
Hi,

I'm trying to compile for MMDVM_POD on STM32F105RBT MCU, but when arrive to RSSIRB.H i have this output:

arm-none-eabi-g++ -MMD -mthumb -mlittle-endian -mcpu=cortex-m3 -Wall -I. -IC:\Users\LAB\Desktop\STM32Cube_FW_F1_V1.6.0\Drivers\CMSIS/Include -IC:\Users\LAB\Desktop\STM32Cube_FW_F1_V1.6.0\Drivers\CMSIS/Device/ST/STM32F1xx/Include -Isystem_stm32f1xx  -DSTM32F105xC -Os -flto -ffunction-sections -fdata-sections -g  -nostdlib -fno-exceptions -fno-rtti -c RSSIRB.cpp -o obj/RSSIRB.o
In file included from RSSIRB.cpp:21:0:
RSSIRB.h:37:20: error: expected ')' before 'length'
   CRSSIRB(uint16_t length);
                    ^
RSSIRB.h:39:3: error: 'uint16_t' does not name a type
   uint16_t getSpace() const;
   ^
RSSIRB.h:41:3: error: 'uint16_t' does not name a type
   uint16_t getData() const;
   ^
RSSIRB.h:43:12: error: 'uint16_t' has not been declared
   bool put(uint16_t rssi);
            ^
RSSIRB.h:45:12: error: 'uint16_t' has not been declared
   bool get(uint16_t& rssi);
            ^
RSSIRB.h:50:3: error: 'uint16_t' does not name a type
   uint16_t           m_length;
   ^
RSSIRB.h:51:12: error: 'uint16_t' does not name a type
   volatile uint16_t* m_rssi;
            ^
RSSIRB.h:52:12: error: 'uint16_t' does not name a type
   volatile uint16_t  m_head;
            ^
RSSIRB.h:53:12: error: 'uint16_t' does not name a type
   volatile uint16_t  m_tail;
            ^
RSSIRB.cpp:23:17: error: expected constructor, destructor, or type conversion before '(' token
 CRSSIRB::CRSSIRB(uint16_t length) :
                 ^
RSSIRB.cpp:34:1: error: 'uint16_t' does not name a type
 uint16_t CRSSIRB::getSpace() const
 ^
RSSIRB.cpp:51:1: error: 'uint16_t' does not name a type
 uint16_t CRSSIRB::getData() const
 ^
RSSIRB.cpp:61:19: error: 'bool CRSSIRB::put' is not a static data member of 'class CRSSIRB'
 bool CRSSIRB::put(uint16_t rssi)
                   ^
RSSIRB.cpp:61:19: error: 'uint16_t' was not declared in this scope
RSSIRB.cpp:62:1: error: expected ',' or ';' before '{' token
 {
 ^
make: *** [obj/RSSIRB.o] Error 1

phl0, 2018-03-04:
What is in your Config.h? And did you clone the STM32F1 lib?

phl0, 2018-09-18:
I tried to reproduce this error without success. However I committed some changes to Makefile.CMSIS but I guess those do not solve the problem. Is the error about RSSIRB.cpp line 62 still appearing?

badtime, 2018-09-18:
Well Florian, at the end i had decided to use linux to compile, and I've not problems anymore.
I dont know what was the issue on windows, but on linux all work fine.

Thank you for the interest, but for me this issue is "almost" solved.

73

phl0, 2018-09-18:
Well then. I'm gonna close this issue now. No fun to install a Windows machine to test that especially since the code has not been changed recently when you found the issue. Feel free to re-open it again.

closed 2018-09-18 by phl0