g4klx/MMDVM #164: Strange Behavier with MMDVM Repeater.

citynetcomputing, 2018-04-15:
Hello Jonathan,

I have a MMDVM PI based DMR full duplex repeater that otherwise works perfectly, The problem is when using an MD380 to talk to my XPR7550e the XPR keeps dropping and you can hear subtle audio loss when it does. I am not sure of this is related to the talker alias issues i'm reading or TX Interrupt, I don't know.. But this is making this MMDVM Repeater utterly useless as were all XPR users.

Talking from one repeater - over the BM link - to the MMDVM repeater works perfectly MD to XPR.
Talking from the MMDVM repeater - over the BM link - to anyone else in the world with a MD works perfectly.

Only issue is talking to someone on my same repeater (via RF) radio to radio shows the issue...

Please help,

73,

citynetcomputing, 2018-04-15:
Just a thought, is there a way to comment out Talker Alias options and recompile the firmware without it?

Thanks!

ct1jib, 2018-04-16:
move DumpTAData=1 to 0, this disable the use of TA

citynetcomputing, 2018-04-16:
Hi,

Thanks but that does not disable TA it just stops it from sending extra data.

Cheers!

juribeparada, 2018-04-18:
Hi, some questions:
- Are you using latest firmware for the MMDVM board?
- latest MMDVMHost?
- latest firmware for MD380?
- have you tried any other Chinese radio (GD77 for example)?

Regards

citynetcomputing, 2018-04-18:
1. Yes
2. Yes
3. No as I have no other except XPR's and 380's but there are many people complaining of my same issue and some are listed here in the "issues" tab.

https://github.com/g4klx/MMDVM/issues/144
https://github.com/g4klx/MMDVM/issues/127

juribeparada, 2018-04-18:
OK, I see. Well, during February (during my vacations, when I have plenty of time...) I worked in this issue, specially the network problem with Motorola radios, and also repeater only problem with Motorola radios. It was fixed, specially the issues pointed by Guillaume VA2NA. I think your problem is slightly different. Unfortunately I don't have access to a XPR radio, I have a DGP8550, that works perfect with MD380 using a MMDVM repeater.
I understand you have a MMDVM-Pi board, I want to know what board is, based on STM32F446, STM32F105, etc.

citynetcomputing, 2018-04-18:
Unfortunately most people around here are MD380 and XPR ;) I tried both the Chinese version of the MMDVM-PI and the Official MMDVM-PI from Bruce and both produce the same result. I have ordered a STM32-DVM PI Hat to test soon as well. Using an official XPR Repeater resolves the issue so its definitely MMDVM Based.

citynetcomputing, 2018-04-18:
Also transmit wise the XPR works perfectly, It receiving is the issue.

juribeparada, 2018-04-18:
Please see here: https://github.com/g4klx/MMDVM/issues/144
and here: https://groups.yahoo.com/neo/groups/mmdvm/conversations/messages/16247

"Also while listening to the MD380 i can hear the discrepancy in the audio when the XPR drops carrier and re-aquires.". That is interesting, then we have some abnormalities in the DMR transmission. This happened with any MMDVM-Pi board, or with the Chinese board?. This small "discrepancy" in MD380 happens when other MD380 is transmitting to the repeater?

citynetcomputing, 2018-04-18:
Both boards. and Yes.

citynetcomputing, 2018-04-18:
Yes, - https://groups.yahoo.com/neo/groups/mmdvm/conversations/messages/16247
This sounds the same. Any fix for this yet??

va2na, 2018-04-18:
Yes the fix have been release for the 2 issues, one for RF to RF problem and one for NEt to RF problem.

Be sure you are using the latest firmware in your modem,

73

Guillaume VA2NA

citynetcomputing, 2018-04-18:
Ok,

Well it hasn't been fixed then cause the problem still lives... lol

va2na, 2018-04-18:
HEre Jay we have around 12 repeater on MMDVM and we have all XPR 6550 and 4550, the 2 issue i have notice and confirm on DSD+ is solved and i am 100% confident about that.

Maybe you can use DSD+ , your Motorola radio, your MD 380, the MMDVM log file in real time and do exactly what i do ...to indentify where and when the lost happen...this how i have work.

I am repeating my self but 2 issues i found this way have been corrected by Andy and all the motortro user on this side is happy now 😃

73

citynetcomputing, 2018-04-18:
I have DSD+ now, How to i connect it to the DM380? im assuming to decode?

va2na, 2018-04-18:
you need a sdr receiver

citynetcomputing, 2018-04-18:
Ok ill post my findings here when done.

Thanks.

citynetcomputing, 2018-04-19:
Ok,

So here is something I found in debug mode.. Audio sequence 2 is repeated. That's when the XPR drops..
See below.. Thoughts? All the remaining sequences in the transmission are just 1-5..

D: 2018-04-19 23:11:43.039 DMR Slot 2, raw RSSI: 3660, reported RSSI: -141 dBm
D: 2018-04-19 23:11:43.039 DMR Slot 2, audio sequence no. 0, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:43.092 DMR Slot 2, audio sequence no. 1, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:43.097 DMR Slot 2, audio sequence no. 2, errs: 6/141 (4.3%)
D: 2018-04-19 23:11:43.154 DMR Slot 2, audio sequence no. 2, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:43.212 DMR Slot 2, audio sequence no. 3, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:43.274 DMR Slot 2, audio sequence no. 4, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:43.332 DMR Slot 2, audio sequence no. 5, errs: 0/141 (0.0%)

Full transmisstion:

M: 2018-04-19 23:11:41.954 Debug: DMRSlotRX: voice sync found slot/pos/centre/threshold 2 449 -270 591
D: 2018-04-19 23:11:41.959 DMR Slot 2, raw RSSI: 3665, reported RSSI: -141 dBm
D: 2018-04-19 23:11:41.960 DMR Slot 2, audio sequence no. 0, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:42.012 DMR Slot 2, audio sequence no. 1, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:42.074 DMR Slot 2, audio sequence no. 2, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:42.132 DMR Slot 2, audio sequence no. 3, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:42.194 DMR Slot 2, audio sequence no. 4, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:42.252 DMR Slot 2, audio sequence no. 5, errs: 0/141 (0.0%)
M: 2018-04-19 23:11:42.314 Debug: DMRSlotRX: voice sync found slot/pos/centre/threshold 2 449 -244 587
D: 2018-04-19 23:11:42.319 DMR Slot 2, raw RSSI: 3662, reported RSSI: -141 dBm
D: 2018-04-19 23:11:42.319 DMR Slot 2, audio sequence no. 0, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:42.372 DMR Slot 2, audio sequence no. 1, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:42.434 DMR Slot 2, audio sequence no. 2, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:42.492 DMR Slot 2, audio sequence no. 3, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:42.554 DMR Slot 2, audio sequence no. 4, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:42.612 DMR Slot 2, audio sequence no. 5, errs: 0/141 (0.0%)
M: 2018-04-19 23:11:42.674 Debug: DMRSlotRX: voice sync found slot/pos/centre/threshold 2 449 -231 583
D: 2018-04-19 23:11:42.679 DMR Slot 2, raw RSSI: 3660, reported RSSI: -141 dBm
D: 2018-04-19 23:11:42.679 DMR Slot 2, audio sequence no. 0, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:42.732 DMR Slot 2, audio sequence no. 1, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:42.794 DMR Slot 2, audio sequence no. 2, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:42.852 DMR Slot 2, audio sequence no. 3, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:42.914 DMR Slot 2, audio sequence no. 4, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:42.972 DMR Slot 2, audio sequence no. 5, errs: 0/141 (0.0%)
M: 2018-04-19 23:11:43.034 Debug: DMRSlotRX: voice sync found slot/pos/centre/threshold 2 449 -227 579
D: 2018-04-19 23:11:43.039 DMR Slot 2, raw RSSI: 3660, reported RSSI: -141 dBm
D: 2018-04-19 23:11:43.039 DMR Slot 2, audio sequence no. 0, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:43.092 DMR Slot 2, audio sequence no. 1, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:43.097 DMR Slot 2, audio sequence no. 2, errs: 6/141 (4.3%)
D: 2018-04-19 23:11:43.154 DMR Slot 2, audio sequence no. 2, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:43.212 DMR Slot 2, audio sequence no. 3, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:43.274 DMR Slot 2, audio sequence no. 4, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:43.332 DMR Slot 2, audio sequence no. 5, errs: 0/141 (0.0%)
M: 2018-04-19 23:11:43.394 Debug: DMRSlotRX: voice sync found slot/pos/centre/threshold 2 449 -231 580
D: 2018-04-19 23:11:43.399 DMR Slot 2, raw RSSI: 3660, reported RSSI: -141 dBm
D: 2018-04-19 23:11:43.400 DMR Slot 2, audio sequence no. 0, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:43.452 DMR Slot 2, audio sequence no. 1, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:43.514 DMR Slot 2, audio sequence no. 2, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:43.572 DMR Slot 2, audio sequence no. 3, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:43.634 DMR Slot 2, audio sequence no. 4, errs: 0/141 (0.0%)
D: 2018-04-19 23:11:43.692 DMR Slot 2, audio sequence no. 5, errs: 0/141 (0.0%)

va2na, 2018-04-19:
You got it!

Andy will do the rest ;)

Guillaume

citynetcomputing, 2018-04-19:
YAYY!!! ;)

Thank you!!

73

juribeparada, 2018-04-20:
Hi,

very interesting finding. I will try to fix during weekend. This sequence 2 repetition happens when you are using MD380?, not with XPR?
What hardware are you using for host computer? (Raspberry Pi, Orange Pi, etc)

Regards,
Andy.

citynetcomputing, 2018-04-20:
Hi Andy,

Yes I typically transmit using an MD380, Listen for the audio discrepancies with another MD380 and also listen and watch the XPR Radios for them to start dropping out. Host is the official MMDVM on a PI-3 with latest MMDVMHost and Second last MMDVM firmware commit.

Is that repeat sequence thing not normal or?

juribeparada, 2018-04-20:
It is not normal, at least I've never seen that sequence repetition before, even using a MD380 + MMDVM Pi board + RPi-3. Something could be different in your system that produces that behaviour. But anyway, it could be easily fixed dropping the repeated frame in MMDVMHost.

citynetcomputing, 2018-04-20:
Ok, Well I do appreciate your time on it!. Wish I knew why it was making that happen lol.

juribeparada, 2018-04-21:
Can you try to recompile your MMDVMHost using the source code from my GitHub?:
https://github.com/juribeparada/MMDVMHost

Please test if this fix work for you or not. If this works, I will merge to official MMDVMHost.

Andy.

citynetcomputing, 2018-04-21:
Sure. Ill do this later this evening and report back. 

Cheers,
Jay

citynetcomputing, 2018-04-21:
Looks like MMDVMHost crashed also..

M: 2018-04-21 22:32:58.247 Debug: Mode set to Idle
M: 2018-04-21 22:32:58.253 Debug: Mode set to DMR
M: 2018-04-21 22:33:01.250 Debug: Mode set to Idle
M: 2018-04-21 22:33:06.249 Closing the MMDVM
I: 2018-04-21 22:33:07.212 Stopped the DMR Id lookup reload thread
M: 2018-04-21 22:33:07.212 DMR, Closing DMR Network
I: 2018-04-21 22:33:07.212 MMDVMHost-20180327 exited on receipt of SIGTERM

juribeparada, 2018-04-21:
Uhmm, that is very strange, here working fine with my setup. Could you copy here more lines of your log?

citynetcomputing, 2018-04-22:
Its the same as what i posted. Just still lists the extra audio sequence.

citynetcomputing, 2018-04-22:
Same thing on raw raspbian with your MMDVMHost install and randomly crashes...

M: 2018-04-22 00:43:57.042 Debug: DMRSlotRX: voice sync found slot/pos/centre/threshold 2 449 -67 134
D: 2018-04-22 00:43:57.047 DMR Slot 2, audio sequence no. 0, errs: 0/141 (0.0%)
D: 2018-04-22 00:43:57.100 DMR Slot 2, audio sequence no. 1, errs: 0/141 (0.0%)
D: 2018-04-22 00:43:57.161 DMR Slot 2, audio sequence no. 2, errs: 0/141 (0.0%)
D: 2018-04-22 00:43:57.220 DMR Slot 2, audio sequence no. 3, errs: 0/141 (0.0%)
D: 2018-04-22 00:43:57.281 DMR Slot 2, audio sequence no. 4, errs: 0/141 (0.0%)
D: 2018-04-22 00:43:57.340 DMR Slot 2, audio sequence no. 5, errs: 0/141 (0.0%)
M: 2018-04-22 00:43:57.402 Debug: DMRSlotRX: voice sync found slot/pos/centre/threshold 2 449 -73 134
D: 2018-04-22 00:43:57.407 DMR Slot 2, audio sequence no. 0, errs: 0/141 (0.0%)
D: 2018-04-22 00:43:57.460 DMR Slot 2, audio sequence no. 1, errs: 0/141 (0.0%)
D: 2018-04-22 00:43:57.521 DMR Slot 2, audio sequence no. 2, errs: 0/141 (0.0%)
D: 2018-04-22 00:43:57.580 DMR Slot 2, audio sequence no. 3, errs: 0/141 (0.0%)
D: 2018-04-22 00:43:57.641 DMR Slot 2, audio sequence no. 4, errs: 0/141 (0.0%)
D: 2018-04-22 00:43:57.700 DMR Slot 2, audio sequence no. 5, errs: 0/141 (0.0%)
M: 2018-04-22 00:43:57.762 Debug: DMRSlotRX: voice terminator found slot/pos/centre/threshold 2 449 -71 134
M: 2018-04-22 00:43:57.768 DMR Slot 2, received RF end of voice transmission, 2.9 seconds, BER: 0.0%
M: 2018-04-22 00:44:00.822 Debug: DMRSlotRX: voice header found slot/pos/centre/threshold 2 449 -96 129
M: 2018-04-22 00:44:00.831 DMR Slot 2, received RF voice header from 3162025 to TG 9
M: 2018-04-22 00:44:00.882 Debug: DMRSlotRX: voice header found slot/pos/centre/threshold 2 449 -90 130
M: 2018-04-22 00:44:00.942 Debug: DMRSlotRX: voice header found slot/pos/centre/threshold 2 449 -84 131
M: 2018-04-22 00:44:01.002 Debug: DMRSlotRX: voice sync found slot/pos/centre/threshold 2 449 -79 132
D: 2018-04-22 00:44:01.007 DMR Slot 2, audio sequence no. 0, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:01.060 DMR Slot 2, audio sequence no. 1, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:01.121 DMR Slot 2, audio sequence no. 2, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:01.180 DMR Slot 2, audio sequence no. 3, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:01.241 DMR Slot 2, audio sequence no. 4, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:01.300 DMR Slot 2, audio sequence no. 5, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:01.305 DMR Slot 2, audio sequence no. 2, errs: 7/141 (5.0%)
M: 2018-04-22 00:44:01.362 Debug: DMRSlotRX: voice sync found slot/pos/centre/threshold 2 449 -73 133
D: 2018-04-22 00:44:01.367 DMR Slot 2, audio sequence no. 0, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:01.420 DMR Slot 2, audio sequence no. 1, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:01.481 DMR Slot 2, audio sequence no. 2, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:01.540 DMR Slot 2, audio sequence no. 3, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:01.601 DMR Slot 2, audio sequence no. 4, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:01.660 DMR Slot 2, audio sequence no. 5, errs: 0/141 (0.0%)
M: 2018-04-22 00:44:01.722 Debug: DMRSlotRX: voice sync found slot/pos/centre/threshold 2 449 -73 133
D: 2018-04-22 00:44:01.727 DMR Slot 2, audio sequence no. 0, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:01.780 DMR Slot 2, audio sequence no. 1, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:01.841 DMR Slot 2, audio sequence no. 2, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:01.900 DMR Slot 2, audio sequence no. 3, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:01.961 DMR Slot 2, audio sequence no. 4, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:02.020 DMR Slot 2, audio sequence no. 5, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:02.071 DMR, Sending authorisation
M: 2018-04-22 00:44:02.082 Debug: DMRSlotRX: voice sync found slot/pos/centre/threshold 2 449 -73 132
D: 2018-04-22 00:44:02.087 DMR Slot 2, audio sequence no. 0, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:02.123 DMR, Sending configuration
D: 2018-04-22 00:44:02.140 DMR Slot 2, audio sequence no. 1, errs: 0/141 (0.0%)
M: 2018-04-22 00:44:02.171 DMR, Logged into the master successfully
D: 2018-04-22 00:44:02.201 DMR Slot 2, audio sequence no. 2, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:02.201 Network Transmitted

citynetcomputing, 2018-04-22:
D: 2018-04-22 00:44:02.201 Network Transmitted
D: 2018-04-22 00:44:02.201 0000:  44 4D 52 44 16 30 3F A9 00 00 09 00 30 3F A9 82    *DMRD.0?.....0?..*
D: 2018-04-22 00:44:02.202 0010:  AF 9D 57 35 A9 E9 A3 16 24 04 23 6A 1D A9 E9 A3    *..W5....$.#j....*
D: 2018-04-22 00:44:02.202 0020:  16 21 70 30 90 60 57 44 06 23 5A 0D A9 E9 A3 16    *.!p0.`WD.#Z.....*
D: 2018-04-22 00:44:02.202 0030:  24 04 22 5A 1F 00 00                               *$."Z...*
D: 2018-04-22 00:44:02.260 DMR Slot 2, audio sequence no. 3, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:02.260 Network Transmitted
D: 2018-04-22 00:44:02.260 0000:  44 4D 52 44 17 30 3F A9 00 00 09 00 30 3F A9 83    *DMRD.0?.....0?..*
D: 2018-04-22 00:44:02.260 0010:  AF 9D 57 35 A9 E9 A3 16 24 04 22 5A 0D A9 E9 A3    *..W5....$."Z....*
D: 2018-04-22 00:44:02.260 0020:  16 21 70 C0 F1 40 A7 44 04 23 7A 0D A9 E9 A3 16    *.!p..@.D.#z.....*
D: 2018-04-22 00:44:02.260 0030:  24 04 23 6A 0F 00 00                               *$.#j...*
D: 2018-04-22 00:44:02.322 DMR Slot 2, audio sequence no. 4, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:02.322 Network Transmitted
D: 2018-04-22 00:44:02.322 0000:  44 4D 52 44 18 30 3F A9 00 00 09 00 30 3F A9 84    *DMRD.0?.....0?..*
D: 2018-04-22 00:44:02.322 0010:  AF 9D 57 35 A9 E9 A3 16 24 04 33 7A 1D A9 E9 A3    *..W5....$.3z....*
D: 2018-04-22 00:44:02.322 0020:  16 21 50 F1 E1 E1 10 74 04 23 7A 1D A9 E9 A3 16    *.!P....t.#z.....*
D: 2018-04-22 00:44:02.322 0030:  24 04 23 7A 1F 00 00                               *$.#z...*
D: 2018-04-22 00:44:02.380 DMR Slot 2, audio sequence no. 5, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:02.380 Network Transmitted
D: 2018-04-22 00:44:02.380 0000:  44 4D 52 44 19 30 3F A9 00 00 09 00 30 3F A9 85    *DMRD.0?.....0?..*
D: 2018-04-22 00:44:02.380 0010:  AF 9D 57 35 A9 E9 A3 16 24 04 23 6A 1D A9 E9 A3    *..W5....$.#j....*
D: 2018-04-22 00:44:02.380 0020:  16 21 10 00 00 00 0E 24 04 23 6A 1D A9 E9 A3 16    *.!.....$.#j.....*
D: 2018-04-22 00:44:02.380 0030:  24 04 23 6A 1F 00 00                               *$.#j...*
M: 2018-04-22 00:44:02.442 Debug: DMRSlotRX: voice sync found slot/pos/centre/threshold 2 449 -71 133
D: 2018-04-22 00:44:02.447 DMR Slot 2, audio sequence no. 0, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:02.447 Network Transmitted
D: 2018-04-22 00:44:02.447 0000:  44 4D 52 44 1A 30 3F A9 00 00 09 00 30 3F A9 90    *DMRD.0?.....0?..*
D: 2018-04-22 00:44:02.447 0010:  AF 9D 57 35 A9 E9 A3 16 24 04 23 6A 0D A9 E9 A3    *..W5....$.#j....*
D: 2018-04-22 00:44:02.447 0020:  16 27 55 FD 7D F7 5F 74 04 22 4A 0D B9 E9 A3 16    *.'U.}._t."J.....*
D: 2018-04-22 00:44:02.447 0030:  24 06 23 4B 0F 00 00                               *$.#K...*
D: 2018-04-22 00:44:02.500 DMR Slot 2, audio sequence no. 1, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:02.500 Network Transmitted
D: 2018-04-22 00:44:02.500 0000:  44 4D 52 44 1B 30 3F A9 00 00 09 00 30 3F A9 81    *DMRD.0?.....0?..*
D: 2018-04-22 00:44:02.500 0010:  AF 9D 57 35 B9 E9 A3 16 24 04 23 6B 0D A9 E9 A3    *..W5....$.#k....*
D: 2018-04-22 00:44:02.500 0020:  16 21 30 30 30 A0 09 14 04 22 4A 0D B9 F9 A3 16    *.!000...."J.....*
D: 2018-04-22 00:44:02.500 0030:  24 04 22 4A 1F 00 00                               *$."J...*
D: 2018-04-22 00:44:02.562 DMR Slot 2, audio sequence no. 2, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:02.562 Network Transmitted
D: 2018-04-22 00:44:02.562 0000:  44 4D 52 44 1C 30 3F A9 00 00 09 00 30 3F A9 82    *DMRD.0?.....0?..*
D: 2018-04-22 00:44:02.562 0010:  AF 9D 57 35 B9 F9 A3 16 24 06 23 4A 1D A9 E9 A3    *..W5....$.#J....*
D: 2018-04-22 00:44:02.562 0020:  16 21 70 30 90 60 57 44 04 23 6A 3F F9 89 E6 10    *.!p0.`WD.#j?....*
D: 2018-04-22 00:44:02.562 0030:  04 2C A3 3B B3 00 00                               *.,.;...*
D: 2018-04-22 00:44:02.620 DMR Slot 2, audio sequence no. 3, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:02.620 Network Transmitted
D: 2018-04-22 00:44:02.620 0000:  44 4D 52 44 1D 30 3F A9 00 00 09 00 30 3F A9 83    *DMRD.0?.....0?..*
D: 2018-04-22 00:44:02.620 0010:  AF 9D 57 35 C5 7E 96 E4 72 09 5F A4 41 EE 47 E6    *..W5.~..r._.A.G.*
D: 2018-04-22 00:44:02.620 0020:  10 21 70 C0 F1 40 A7 43 6B 42 4C 59 DF 8C E1 32    *.!p..@.CkBLY...2*
D: 2018-04-22 00:44:02.620 0030:  65 B5 D0 44 7F 00 00                               *e..D...*
D: 2018-04-22 00:44:02.682 DMR Slot 2, audio sequence no. 4, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:02.682 Network Transmitted
D: 2018-04-22 00:44:02.682 0000:  44 4D 52 44 1E 30 3F A9 00 00 09 00 30 3F A9 84    *DMRD.0?.....0?..*
D: 2018-04-22 00:44:02.682 0010:  AF 9D 57 35 90 64 A7 94 ED E2 4C 98 00 80 A8 AC    *..W5.d....L.....*
D: 2018-04-22 00:44:02.682 0020:  44 21 50 F1 E1 E1 10 75 5B 32 A7 42 C5 36 C6 D5    *D!P....u[2.B.6..*
D: 2018-04-22 00:44:02.682 0030:  44 A8 D5 35 B3 00 00                               *D..5...*
D: 2018-04-22 00:44:02.740 DMR Slot 2, audio sequence no. 5, errs: 0/141 (0.0%)
D: 2018-04-22 00:44:02.740 Network Transmitted
D: 2018-04-22 00:44:02.740 0000:  44 4D 52 44 1F 30 3F A9 00 00 09 00 30 3F A9 85    *DMRD.0?.....0?..*
D: 2018-04-22 00:44:02.740 0010:  AF 9D 57 35 40 25 98 48 02 15 AF 6B 72 9D CD A6    *..W5@%.H...kr...*
D: 2018-04-22 00:44:02.740 0020:  52 01 10 00 00 00 0E 20 9B 05 55 C5 9D CD B6 52    *R...... ..U....R*
D: 2018-04-22 00:44:02.740 0030:  00 99 05 45 C7 00 00                               *...E...*
M: 2018-04-22 00:44:02.802 Debug: DMRSlotRX: voice terminator found slot/pos/centre/threshold 2 449 -64 133
[... cut ...]
D: 2018-04-22 00:44:21.579 0010:  00 00 00 00 9B 23 84 17 62 ED A7 37 4D 88 00 A4    *.....#..b..7M...*
D: 2018-04-22 00:44:21.579 0020:  75 01 57 24 7B EA 50 72 DB B4 07 0F F2 9C C2 70    *u.W${.Pr.......p*
D: 2018-04-22 00:44:21.579 0030:  26 8D 2D 83 B3 00 00                               *&.-....*
^CM: 2018-04-22 00:44:21.599 Closing the MMDVM
I: 2018-04-22 00:44:22.026 Stopped the DMR Id lookup reload thread
M: 2018-04-22 00:44:22.027 DMR, Closing DMR Network
I: 2018-04-22 00:44:22.027 MMDVMHost-20180327 exited on receipt of SIGINT

citynetcomputing, 2018-04-22:
Im just doing a git clone, make and copying the MMDVMHost binary over... anything else special I should be doing?

juribeparada, 2018-04-22:
You have to copy the binary, nothing more. Well, at least I don't see any consecutive audio sequence repetition (your first log showed two "audio sequence no. 2"), but now I see an extra "audio sequence no. 2", uhmmm. It's really very strange. It might be a hardware problem, is it possible to test with a different computer board?. I will continue to think about this tomorrow.

citynetcomputing, 2018-04-22:
Ive tested with three different pi3's and two different MMDVM's..

citynetcomputing, 2018-04-22:
Yes the "D: 2018-04-21 22:21:00.942 DMR Slot 2, audio sequence no. 2, errs: 6/141 (4.3%)" are still coming through..

juribeparada, 2018-04-22:
Hi, try again from my GitHub:
https://github.com/juribeparada/MMDVMHost

citynetcomputing, 2018-04-22:
Hello,
Haha, I actually just did it before you asked... version 20180327 git #28f8d9e? correct?

Still same issue.. and its always sequence number 2 for some reason...

D: 2018-04-22 01:18:56.480 DMR Slot 2, audio sequence no. 4, errs: 1/141 (0.7%)
D: 2018-04-22 01:18:56.538 DMR Slot 2, audio sequence no. 5, errs: 1/141 (0.7%)
M: 2018-04-22 01:18:56.600 Debug: DMRSlotRX: voice sync found slot/pos/centre/threshold 2 451 -73 145
D: 2018-04-22 01:18:56.605 DMR Slot 2, audio sequence no. 0, errs: 1/141 (0.7%)
D: 2018-04-22 01:18:56.658 DMR Slot 2, audio sequence no. 1, errs: 0/141 (0.0%)
D: 2018-04-22 01:18:56.719 DMR Slot 2, audio sequence no. 2, errs: 0/141 (0.0%)
D: 2018-04-22 01:18:56.778 DMR Slot 2, audio sequence no. 3, errs: 1/141 (0.7%)
D: 2018-04-22 01:18:56.839 DMR Slot 2, audio sequence no. 4, errs: 1/141 (0.7%)
D: 2018-04-22 01:18:56.898 DMR Slot 2, audio sequence no. 5, errs: 0/141 (0.0%)
D: 2018-04-22 01:18:56.903 DMR Slot 2, audio sequence no. 2, errs: 6/141 (4.3%)
M: 2018-04-22 01:18:56.960 Debug: DMRSlotRX: voice sync found slot/pos/centre/threshold 2 451 -78 144
D: 2018-04-22 01:18:56.965 DMR Slot 2, audio sequence no. 0, errs: 0/141 (0.0%)
D: 2018-04-22 01:18:57.018 DMR Slot 2, audio sequence no. 1, errs: 1/141 (0.7%)
D: 2018-04-22 01:18:57.079 DMR Slot 2, audio sequence no. 2, errs: 2/141 (1.4%)
D: 2018-04-22 01:18:57.138 DMR Slot 2, audio sequence no. 3, errs: 1/141 (0.7%)

juribeparada, 2018-04-22:
Nop, git #c91b7f7, and I didn't the date, I've just done 5 min before

citynetcomputing, 2018-04-22:
lol. standby..

citynetcomputing, 2018-04-22:
Well sir, i think you got it that time. Dont appear to be doing it now. What did you do? ;)

juribeparada, 2018-04-22:
Now that code is dropping non-consecutive frames and repeated frames. I want to think this code better to be sure this will not degrade DMR performance, until to merge to oficial MMDVMHost. During the meantime, please test more extensively and tell me if this really solve the drops with XPR radios. Anyway, the root problem is not solved, it seems to me a serial port issue.

citynetcomputing, 2018-04-22:
Sounds good. Will do and let me know if you need anything in the mean time.

citynetcomputing, 2018-05-09:
Hi Andy,

I tried to report back but cannot find my issue in the tracker. Everything seems to still be working ok. Was that version committed?

73,

citynetcomputing, 2018-05-09:
Found it ;)

closed 2018-09-18 by phl0