g4klx/MMDVM #166: IPv6 Support

citynetcomputing, 2018-04-23:
Andy,

Any thoughts or plans on IPv6 Support?

Thanks!

juribeparada, 2018-04-23:
Not from my side at least. Network socket implementation was done by Jonathan G4KLX. Why do you need IPv6 support?

Andy.

citynetcomputing, 2018-04-23:
Well because in the US in the coming years we will be all IPv6 only as we have run out of IPv4 addresses, I have an all IPv6 multi-site Management network and the BM Masters support IPv6. Right now with the limited use of IPv6 it would make that transport quite uncongested unlike IPv4 is currently ;)

And I like IPv6.. ;)

alistairmackenzie, 2018-10-01:
From what I can tell this is not an issue with MMDVM but perhaps an enhancement for [MMDVMHost](https://github.com/g4klx/MMDVMHost).

I'm also quite interested in supporting IPv6 but the brandmeister master servers will need to support it too. Unfortunately, this does not seem to be the case at the moment otherwise I would add the support in to MMDVMHost.

citynetcomputing, 2018-10-01:
Hello alistairmackenzie,

 Yes, All of the BM Masters are IPv6 enabled. Rudy the BM admin mentioned it to me which is why I initially asked. ;)

alistairmackenzie, 2018-10-01:
That's great to hear. I'm not seeing any addresses on the hostnames for the server I use here, UK and Ireland. Obviously, these would be needed before IPv6 is useful.

Perhaps I'm missing something and you can point me in the right direction.

citynetcomputing, 2018-10-01:
ping -6 3101.repeater.net

Pinging 3101.repeater.net [2604:180:2:1593::7c53] with 32 bytes of data.

citynetcomputing, 2018-10-01:
Your servers may not be ipv6 compliant. In the US here we are.

alistairmackenzie, 2018-10-01:
There is a chance it has not been configured on some of the masters DNS records, perhaps not even given an address.

I'll see what I can do with that US one.