g4klx/MMDVM #181: MMDVM repeater get partial voice echo back after each tx

dd5xl, 2018-07-11:
This is a copy of https://github.com/g4klx/MMDVMHost/issues/432 by @shawnchain :
We have exactly the same problem at our local repeater and it definitely belongs to the latest MMDVM Due firmware versions:

With latest commit https://github.com/g4klx/MMDVM/commit/f7056561789665c16e58839107e42a89cb44f03e after every 2nd our 3rd transmission end there is a approx. 1s repetition of the last words spoken echoed back by the repeater. From my feeling the length of the echo seems to be dependent from the length of the transmission: Short transmissions have short echoes (if any), longer transmission have longer trailing echoes.
With commit https://github.com/g4klx/MMDVM/commit/41946ff0aa929d13d30d2c446f63f6bfe8e6d1cc from version bump March 27th the problem can be cured at our repeater, no more echoes at all.

Could it probably belong to the increased ring buffer sizes as introduced with https://github.com/g4klx/MMDVM/commit/c11ff6bf20355c18800c9a433acddc40276d335b ?

Our Setup: RK3066 ARM Linux board, Linux 3.0.36, MMDVMHost https://github.com/g4klx/MMDVMHost/commit/177a69f547bce0dd0b4614875598c8af76302ac as compiled of today.
Arduino Due clone, SP8NTH filter board 13.824 MHz TCXO, 2x T7f 9k6 data transceivers.

There are no visible errors of warnings in the log.
Deactivating all unused modes except DMR in MMDVM.h doesn't help either.

73 de Bert

DL4OCH, 2018-07-16:
Same applies here for my repeater. 
Sometimes I'll get echos of my transmission from milliseconds up to 1-2 seconds in DMR-mode.

Hardware: Pi3, STM32-modem, latest Host and firmware.
Enabled modes: DMR, YSF, CPU-load 8% max.

dd5xl, 2018-07-19:
Problem has been narrowed down to
https://github.com/dd5xl/MMDVM/commit/c11ff6bf20355c18800c9a433acddc40276d335b
where in SerialRB.h ine 35 the ring buffer size has been increased from 370ms to 2000ms.
Reverting this change cures the echo problem in a way that echoes don't annoy any more, however the effect is still noticable sometimes.

dd5xl, 2018-07-19:
@juribeparada 
Hi Andy, could you please check your commit as above and advise why this has been increased that much?
Tnx es 73 de Bert

juribeparada, 2018-07-19:
Hi, this was necessary to stop DMR TX buffer overflow, which produces drops in the audio, and very ugly drops in some specific radios. This is very notorious when you have too much delay in the network. If we reduce the buffer, you will have people complaining about his again. Maybe a value of 1000 could solve this issue and reduce your echo, but it is obvious that this issue requires more work to solve both problems. You must be patience to wait for a better solution, remember this is done during the free time of the developers.

dd5xl, 2018-07-19:
:-) Yes, same here....
I can imagine there was a reason for increasing the buffer and I agree, there must be a different root cause for the mentioned delay, the increased buffer is just emphasizing the symptom, but is not the main issue.
I think we'll need somebody more familiar with the code and the mechanisms behind to investigate this issue successfully... ;-)

va2na, 2018-10-01:
Get same problem here!

dd5xl, 2018-10-03:
I've fixed the issue locally by setting the ringbuffer size in SerialRB.h to 370ms for SAM3X boards too:

Line 35:
`#if defined(STM32F105xC) || defined(__MK20DX256__) || defined(__SAM3X8E__)`

juribeparada, 2018-10-08:
Finally I decided to restore ring buffer value to 370. I don't have much time to investigate this better for now, and seems this change solves the problem.

closed 2018-11-08 by juribeparada