g4klx/MMDVM #187: #define MODE_LEDS and // #define MODE_PINS instead #define ARDUINO_MODE_PINS

badtime, 2018-08-05:
Hi all,

I'm trying to compile for STM32F105 (MMDVM_POG), but in the older release i uncomment #define ARDUINO_MODE_PINS , #define STM32F1_POG , the right clock, and all work fine, but in the latest release i find #define MODE_LEDS and // #define MODE_PINS , I have tryed to uncomment once or all, but when i flash the hex (flash is 100% done) the MMDVM_POG stop to work, no initial led staging, only PWR and LED still powered up.
What am I doing wrong?

phl0, 2018-08-05:
It should be 

   #define STM32F1_POG
   #define MODE_LEDS

only. I am not sure if also mode pins for the F1 are defined.

phl0, 2018-08-05:
Just checked. The STM32F1_POG code does not know about MODE_PINS. So you should be using the MODE_LEDS only.

badtime, 2018-08-05:
I have tryed any combination, and also this configuration (MODE_LEDS STM32F1_POG) but the board seems bad flashed.
Isn't the first time that i flash this board, but is the first time that I have this kind of issue.
The board is ok, because if i use an old release (about march version), back to work properly.
I dont know what is changed since this new releases, but all seem ok, no errors in compiling, no error in flash, but nothing works!
At this point.........I'm lost....

phl0, 2018-08-05:
Can you name the git commit that broke the code?

badtime, 2018-08-05:
So, I compile and flash this board every 1-2 months, is a big gap to determine when something went wrong in the code unfortunately. But the last time that I compiled the code I'm sure "ARDUINO_MODE_PINS" was present, and I uncomment that according to this compiling instruction https://github.com/wojciechk8/MMDVM_pog and it always worked.

badtime, 2018-08-06:
the oldest firmware that i have compiled and don't work is Commits on Jul 5, 2018.
How i can test old firmware before every commit?

phl0, 2018-08-06:
Checkout the older commits and compile :)

badtime, 2018-08-06:
The last working code is commit: e1b99f6ee4968cdab66ca8942c85c3a66266c091 May 14 2018

Starting from commit c11ff6bf20355c18800c9a433acddc40276d335b May 16 2018 the board stop to work.

Can you check and fix this issue please?

I have used always the same configuration:

`// Allow for the use of high quality external clock oscillators
// The number is the frequency of the oscillator in Hertz.
//
// The frequency of the TCXO must be an integer multiple of 48000.
// Frequencies such as 12.0 Mhz (48000 * 250) and 14.4 Mhz (48000 * 300) are suitable.
// Frequencies such as 10.0 Mhz (48000 * 208.333) or 20 Mhz (48000 * 416.666) are not suitable.
//
// For 12 MHz
// #define EXTERNAL_OSC 12000000
// For 12.288 MHz
// #define EXTERNAL_OSC 12288000
// For 14.4 MHz
// #define EXTERNAL_OSC 14400000
// For 19.2 MHz
#define EXTERNAL_OSC 19200000

// Allow the use of the COS line to lockout the modem
// #define USE_COS_AS_LOCKOUT

// Use pins to output the current mode
#define ARDUINO_MODE_PINS

// For the original Arduino Due pin layout
// #define ARDUINO_DUE_PAPA

// For the ZUM V1.0 and V1.0.1 boards pin layout
// #define ARDUINO_DUE_ZUM_V10

// For the SQ6POG board
#define STM32F1_POG

// For the SP8NTH board
// #define ARDUINO_DUE_NTH

// For ST Nucleo-64 STM32F446RE board
// #define STM32F4_NUCLEO_MORPHO_HEADER
// #define STM32F4_NUCLEO_ARDUINO_HEADER

// Use separate mode pins to switch external filters/bandwidth for example
// #define STM32F4_NUCLEO_MODE_PINS

// Pass RSSI information to the host
#define SEND_RSSI_DATA

// Use the modem as a serial repeater for Nextion displays
// #define SERIAL_REPEATER

// To reduce CPU load, you can remove the DC blocker by commenting out the next line
#define USE_DCBLOCKER

#endif
`

g4klx, 2018-08-06:
> The last working code is commit: e1b99f6 May 14 2018
>
> Starting from commit c11ff6b May 16 2018 the board stop to work.

I think ARDUINO_MODE_PINS is now MODE_PINS.

juribeparada, 2018-08-06:
If the problem began from c11ff6b, then there is a memory issue with STM32F105. I will fix this soon, reducing buffers again only for F105.

badtime, 2018-08-06:
Thank you, tell me when i can try if all work done.

juribeparada, 2018-08-06:
Please try now, it should be fixed

badtime, 2018-08-06:
Now the board startup done, but..........does not work anymore with MMDVMHost.
In the LOG give:
M: 2018-08-06 19:26:21.664 Opening the MMDVM
I: 2018-08-06 19:26:23.675 MMDVM protocol version: 1, description: MMDVM 20180723 (D-Star/DMR/System Fusion/P25/NXDN/POCSAG) 19.2000 Mhz (Build: 21:13:53 Aug  6 2018)
E: 2018-08-06 19:26:23.695 Received a NAK to the SET_CONFIG command from the modem

badtime, 2018-08-06:
Ok, now all work well, during the last test i was using an old PI-STAR version and the MMDVMHost give me a NAK error, but after an upgrade all goes fine! this issue can be closed for me, Thanks to all.

closed 2018-08-06 by phl0