g4klx/MMDVM #200: Make -f Makefile.CMSIS fails - missing stm32f1xx.h

rogerclarkmelbourne, 2018-12-11:
I'm trying to build for the STM32F105 target, and presume that I need to use the Makefile.CMSIS for this, since the "Makefile" seems to build for the F4 and F7

However make -f Makefile.CMSIS fails

> 
> rm-none-eabi-g++ -MMD -mthumb -mlittle-endian -mcpu=cortex-m3 -Wall -I. -I/opt/STM32Cube_FW_F1_V1.4.0/Drivers/CMSIS/Include -I/opt/STM32Cube_FW_F1_V1.4.0/Drive
> rs/CMSIS/Device/ST/STM32F1xx/Include -Isystem_stm32f1xx  -DSTM32F105xC -DMADEBYMAKEFILE -Os -flto -ffunction-sections -fdata-sections -g  -nostdlib -fno-excepti
> ons -fno-rtti -c CalDMR.cpp -o obj/CalDMR.o
> In file included from CalDMR.cpp:21:0:
> Globals.h:27:23: fatal error: stm32f1xx.h: No such file or directory
>  #include "stm32f1xx.h"
>                        ^
> compilation terminated.

I can't see find stm32f1xx.h in this repo, and I have using git submodule init and git submodule update to clone / pull the submodules

I guess I must be doing something wrong, but I'm not sure what

phl0, 2018-12-11:
I guess this should help out:
https://github.com/wojciechk8/MMDVM_pog

rogerclarkmelbourne, 2018-12-11:
Thanks

I just this minute found that site.

It says I need the STM32F1 CMSIS from STM's "Cube"

I think I have the Cube installed already, but I will probably download the F1 "package" and manually extract the CMSIS

rogerclarkmelbourne, 2018-12-11:
OK.

I managed to get it to build by downloading the F1 Cube package

Ideally that information needs to be documented somewhere in this repo, because even the information you linked to

https://github.com/wojciechk8/MMDVM_pog

Is not self explanatory. 

I had to look in the F4 and F7 dirs to get an idea of which files were needed from the F1 package, since the STM Cube is now mainly for the HAL rather than the old Standard Peripheral Library etc

closed 2018-12-11 by rogerclarkmelbourne

shawnchain, 2019-05-13:
Same issue here. 
I noticed @juribeparada had modified STM32F4XX/7XX library a little bit for the MMDVM firmware, so I would like to know if we need to modify the STM32F10X as well to build with STM32F105 ?

mytechguyri, 2020-03-22:
Geez, can't we get clear simple instructions.... I just want to build the damn thing.. not hunt around half the internet for a file that seems to be non-existant.... If I need to download something just put in the README   (wget http://youll_need_this_put_it_here)   why does it have to be so complicated... If I clone your git repository, I expect everything I need to build will be there, Make -f Makefile.CMSIS and it should build... not this nonsense

n file included from CalDStarRX.cpp:20:0:
Globals.h:25:10: fatal error: stm32f1xx.h: No such file or directory
 #include "stm32f1xx.h"
          ^~~~~~~~~~~~~
compilation terminated.

SA7BNT, 2020-03-22:
> Geez, can't we get clear simple instructions.... I just want to build the damn thing.. not hunt around half the internet for a file that seems to be non-existant.... If I need to download something [...]

Why be so angry ?
Just use the link which is placed under .gitmodules [MMDVM](https://github.com/g4klx/MMDVM/blob/master/.gitmodules) make a git clone in MMDVM folder. Just tested it without issues.
Have a nice day.

mytechguyri, 2020-03-22:
Just frustrated from searching for the answer... Apologies.

And after all that... The damn board won't take an update..."failed to
initialize".  Should have known better than to buy cheap Chinese junk...