g4klx/MMDVM #205: Zum Radio MMDVM v1.0 with STM32F722 MPU - Fails to init in d6d0315

compsci282, 2019-05-27:
In commit d6d0315 and onward, it appears the compile and flash process for this device is broken.  The status light and other LEDs do not appear to indicate the typical init sequence and the device is no longer recognized by MMDVMHost.

Errors from MMDVMHost:
E: 2019-05-26 19:45:31.065 Unable to read the firmware version after six attempts

Commits prior to this, the hardware status light flashes and the device operates properly and is again recognized by MMDVMHost.

The compile and flash process I used is located here: https://github.com/juribeparada/MMDVM_man/blob/master/MMDVM-Pi_en.md

I substituted:
- "make pi" with "make pi-f722"
- "sudo make deploy-pi" with "sudo make deploy-pi-f7"
- additionally have to run "make deploy-pi-f7" without the sudo as the prior command does place the device into a firmware update mode, but fails to flash with "Got NACK from device on command 0x01" error.  Running for the 2nd time without sudo avoids the reset procedure and then the stm32flash process succeeds.  Not sure if this is a Pi-Star specific issue or another build/F7 issue.

vk7hse, 2019-06-05:
The version of stm32flash thats used does not include support for the STM32F7xx chip this is the cause of the problem.

compsci282, 2019-06-08:
I'm using stm32flash version 0.5 which appears to be the latest version available.

I just tried to update to dfad6d8 tonight and it appears all is well.  It looks like this issue is being discussed and resolved under the d6d0315 commit in comments.  Closing this ticket.

closed 2019-06-08 by compsci282

kf4eok, 2020-04-20:
I have the ZUM Radio MMDVM-Pi Board v0.9 from HRO with firmware 20170501.  I've been following the firmware update guide found here: https://hvdnnotebook.blogspot.com/2018/08/mmdvm-firmware-why-update-how-to-update.html

I keep getting this error:
Using Parser : Raw BINARY
Interface serial_posix: 57600 8E1
Failed to init device.

Is there something I need to do to get the board into firmware write or boot mode?

help, kf4eok@gmail.com
if I sent this to someone can they update them for me. I have two of these boards.
thanks

compsci282, 2020-04-20:
Hi Daniel,
  The instructions you refer to are related to "HS" or HotSpot boards that have an integrated transceiver and antenna(s) on the board.  In the case of the Zum Radio board, the radio elements are external and the MMDVM board is purely a modem board. 

For the Zum Radio board, the process is a bit different.  Take a look at https://github.com/juribeparada/MMDVM_man/blob/master/MMDVM-Pi_v1_0_en.md

Hopefully these instructions help out.  ;-)

kf4eok, 2020-04-22:
Thanks, that did the job.