g4klx/MMDVM #208: MMDVM_POG STMF105 warning while compiling

badtime, 2019-05-31:
Hi,

when i compile MMDVM for STM32F105 /MMDVM_POG) i have a warning on SerialSTM_CMSIS.cpp

in the past i never had this king of warning.

SerialSTM_CMSIS.cpp: In member function 'void CSerialPort::beginInt(uint8_t, int)':
SerialSTM_CMSIS.cpp:121:61: warning: format '%X' expects argument of type 'unsigned int', but argument 5 has type 'uint32_t {aka                                                                               long unsigned int}' [-Wformat=]
    ::sprintf(UDID, "%04X%04X%08X%08X", *id00,*id01,*id1,*id2);
                                                             ^
SerialSTM_CMSIS.cpp:121:61: warning: format '%X' expects argument of type 'unsigned int', but argument 6 has type 'uint32_t {aka                                                                               long unsigned int}' [-Wformat=]
SerialSTM_CMSIS.cpp:121:61: warning: format '%X' expects argument of type 'unsigned int', but argument 5 has type 'uint32_t {aka                                                                               long unsigned int}' [-Wformat=]
SerialSTM_CMSIS.cpp:121:61: warning: format '%X' expects argument of type 'unsigned int', but argument 6 has type 'uint32_t {aka                                                                               long unsigned int}' [-Wformat=]
/tmp/cc7BgDbc.s: Assembler messages:
/tmp/cc7BgDbc.s:777: Warning: ignoring changed section attributes for .data

shawnchain, 2019-06-10:
This compiler warning has been fixed,  mean while you can safely ignore this warning before the PR is get accepted and merged.

shawnchain, 2019-06-10:
please report any issue related to the POGF105 board issue here, I would like to help.

badtime, 2019-06-11:
Thank you Shawn,

thah warning is disappeared, but i have this other warning at the end:

`/tmp/ccV6gYFc.s: Assembler messages:
/tmp/ccV6gYFc.s:17: Warning: ignoring changed section attributes for .data
/tmp/cc9DWTjS.s: Assembler messages:
/tmp/cc9DWTjS.s:98: Warning: ignoring changed section attributes for .data
arm-none-eabi-objdump -S bin/mmdvm.elf > bin/mmdvm.dis
arm-none-eabi-objcopy -O ihex bin/mmdvm.elf bin/mmdvm.hex
arm-none-eabi-size bin/mmdvm.elf
   text    data     bss     dec     hex filename
  51416    2408   42376   96200   177c8 bin/mmdvm.elf`

Anyway the firmware is correctly compiled, though i have not flashed it to test on a board.

Otherwise, talking about POGF105, in the past i have tried to enable this two option in Config file:

`// Use the YSF and P25 LEDs for NXDN
 #define USE_ALTERNATE_NXDN_LEDS

// Use the D-Star and DMR LEDs for POCSAG
 #define USE_ALTERNATE_POCSAG_LEDS`

but instead using two led to define a new non hardware specified mode on the PCB, it stop to power up any mode led! Why? Is a bug?

Thank you 73 IZ8GUR