g4klx/MMDVM #210: Radio Link Discord to HT using MMDVM hotspot

Neoakb, 2019-07-03:
I would like to be able to use my HT to connect to a Discord voice channel. I figured the hotspot could be helpful in this regard. I don't want to link the discord to any existing DMR networks, but have discord on the ras pi zero I have, send that audio through the airwaves to my HT, and be able to use the PTT on my HT to reply. Instead of it being a radio network hotspot, it's a Discord hotspot. 

Might not be legal, might not be possible. I'm just asking around and seeing what comes up.
Thank you

gbelbe, 2020-03-16:
Looks awesome, did you have any ideas of how to do it?

Neoakb, 2020-03-17:
I did not. It was just a random idea I had.