g4klx/MMDVM #215: MMDVM_POG STM32F105 stop working TX after 175 second

sp2ong, 2019-10-25:
Hi,

We have use board MMDVM_POG STM32F105 with MMDVM firmware 20190130 version and lastest MMDVMHost on our repeater. After every 175 second transmission in DMR mode, the MMDVM_POG board is frozen. PTT is ON, COS led is light. Only Power off/on help to works  MMDVM_POG again. 

In MMDVM.ini the TIMEOUT is value 240 seconds.

If we set the timeout value lower than 175 s for example 160 the timeout works and MMDVM_POG board not frozen and we can start new transmission.

Does anybody meet a similar problem?

sp2gch, 2019-10-27:
Yes, I have exactly the same problem and just turning off the power and turning on helps until the same time.

sp2ong, 2019-12-11:
I suppose that it is problem TCXO

closed 2019-12-11 by sp2ong