g4klx/MMDVM #216: After update of several Icom TRX the beep coming from the repeater ok is coming late

hb9gfx, 2019-11-14:
Icom released new firmware for several radios like IC9700 (Firmware 1.20) , ID4100 (Firmware E102), ID 51 Plus (Firmware E2), ID-5100 (Firmware E4) and maybe others.
WIth the update Icom delivered a correction to send DV Data and other minor changes. Since this update with MMDVM the TRX is generating the beep at least 2 sec. later than the MMDVM gives the EOT ok.
This happens with all listed TRX after the update. I checked with several repeaters where I know the hardware behind the repeaters. It happens not with Icom and DVRPTR repeaters. It seem to be a bug in the MMDVM Firmware that the Icom TRX after correcting the firmware in the Icoms is not sending the correct EOT signal after releasing PTT.
 
This is very bad and the TRX is more or less useless, because if a Station is starting to TX you miss the beginn of the TX season. THis is noticed in defracmenting . As soon as the beep from is done the RX of the TRX is ok.
In the 2 videos you can see and listen the issue. Look to the SMeter and listen when the beep is coming.
http://upload.hb9gfx.ch/video-1573685674.mp4 --> ID-4100
http://upload.hb9gfx.ch/video-1573685833.mp4 --ID-51 plus

I made some crosschecks. The ID51 in the video is modifed wih the old firmware to compare.

g4klx, 2019-11-14:
Firstly your logic is incorrect. The fact that the MMDVM follows the Icom D-Star specification to the letter says to me that Icom may have made the change on purpose. The fact that Icom hardware has been working correctly with the MMDVM since 2016 indicates that they have made a bad change, the MMDVM hasn't changed.

The 2 second delay means that the Icom radios have not decoded the end of transmission marker from the MMDVM.  We need to analyse the differences between an Icom/DV-RPTR system and an MMDVM, it's probably something really silly, but indicative of an ulterior motive at play.

g7npw, 2019-11-14:
A friend reported this to me also after he updated his 9700, I suspected it was just the 9700 at the time and maybe some further group delay or something Icom had introduced. I didn’t investigate as my older DStar radios are fine. 

He just said the beep is now a couple of seconds late and he misses the reflector announcement also.
--
Dominic

hb9gfx, 2019-11-14:
> Firstly your logic is incorrect. The fact that the MMDVM follows the Icom D-Star specification to the letter says to me that Icom may have made the change on purpose. The fact that Icom hardware has [...]

I know , all my Icoms worked proper before the update and I downgraded my ID51 plus already . But Icom is not populating the E101 Frimware. I wrote a mail to Icom with no success, I spoke to Ray Novak he declines because it is third party software and not Icom who is doing a problem :-( . Now I did a phone call to Icom Europe and we will see what they will send back if they send something back but until this is not fixed my ID-4100 is useless and makes no fun to work with

hb9gfx, 2019-11-14:
Johnathan, EOT is identified by MMDVM.
M: 2019-11-14 20:01:48.532 D-Star, received RF header from HB9GFX  /4100 to CQCQCQ
M: 2019-11-14 20:02:13.874 D-Star, received RF end of transmission, 25.5 seconds, BER: 0.0%, RSSI: -49/-47/-47 dBm

hb9gfx, 2019-11-15:
I created 2 videos to show the issue.

This happens with all Icom TRX with Firmware from November 2019 for ID4100/5100/31plus/51Plus/51plus2 and Firmware 1.20 in IC9700

http://upload.hb9gfx.ch/ID-4100%20RX%20until%20beep%20at%20the%20end.mp4
http://upload.hb9gfx.ch/ID-4100%20RX%20until%20beep%20at%20the%20end.mp4

hb9gfx, 2019-12-05:
Icom USA (Ray Novak) told me that today 5th Dec. there will be a fix for the Firmware Updates of ID4100/5100/31plus/51Plus/51plus2 and Firmware 1.20 in IC9700 to prevent the delay of the Roger Beep in DStar. 
With the latest firmware, the transceiver sounds the standby beep with no delay, even when the transceiver communicates with other manufacturers' DV equipment.

So lets hope the problem will be solved.

closed 2019-12-05 by hb9gfx