g4klx/MMDVM #218: dpns.dutch-dstar.eu not reachable and significant retry logic

dsegantx, 2019-12-21:
Recently refined my pi-hole setup in my home network and after a day saw hits from my pi-star hitting this domain dpns.dutch-dstar.eu.  After posting on the pi-star forum https://forum.pistar.uk/viewtopic.php?f=3&t=2304 I learned that the domain isn’t responding and MMDVM continues to retry every minute.

The domain owners main site indicates they have moved.  I’m not clear who/how the port/service is supported but it is dead and has been for some time.

Does this code need to find another source for the service or just deprecate something?  
I’ll try to reach out to the website above and see if they have any guidance but wondering if maybe this code base needs to adapt.

Thanks.
-Dave

g4klx, 2020-01-03:
This is actually an ircDDBGateway question rather than an MMDVM one. I have now removed the old Dutch Star URL from the source code for ircDDB Gateway so it should be a little happier now.

dsegantx, 2020-01-07:
Ah thank you for the clarification.  And thank you for fixing it!

closed 2020-01-07 by dsegantx