g4klx/MMDVM #226: Deploy fails to MMDVM-Pi v1.0

m0vse, 2020-04-24:
Hi,

Since commit f28a22802d24de0c2dd4d023be68d757f018e83b deploy to my MMDVM-Pi fails when trying to upload the firmware.

It appears that this is caused by the -c parameter to stm32flash (resume the connection). If I remove -c then it uploads fine but i'm not sure if this will break uploads to other F7xx based boards so I haven't submitted a PR.

Previously (before the above commit) it called stm32ld first which I assume opened the connection but if this failed to upload (because the CPU is not supported by stm32ld) then stm32flash would still upload as there was an existing connection to the CPU open but by removing the call to stm32ld, there is no existing connection to the CPU. Removing -c forces it to start a new connection.

73 Phil

m0vse, 2020-04-26:
For the avoidance of doubt, this is only for builds relating to Cortex-M7 (stm32f72xx) based boards.

g4klx, 2020-04-26:
Please submit a PR, we'll see what happens.

juribeparada, 2020-11-11:
I think removing the "-c" is safe, actually the use of stm32ld was an ugly workaround that I did long time ago. I tested and works fine for deploy in F4XX and F7XX RPi boards. I will add those mods.