g4klx/MMDVM #234: compilation issue for arduino due

F4IGV, 2020-04-26:
hello,
i'm trying to use MMDVM firmware in Arduino DUE with zum radio for mmdvm DUE.
I have rebuild plateform.txt and instaling sam like in build.txt folder.

when i'm trying to compilate MMDVM i have this message:

Arduino: 1.6.9 Hourly Build 2016/04/15 10:19 (Linux), Board: "Arduino Due (Programming Port)"

sketch/DMRDMOTX.cpp.o: In function `CDMRDMOTX::writeByte(unsigned char)':
sketch/DMRDMOTX.cpp:146: undefined reference to `arm_fir_interpolate_q15'
sketch/DMRTX.cpp.o: In function `CDMRTX::writeByte(unsigned char, unsigned char)':
sketch/DMRTX.cpp:277: undefined reference to `arm_fir_interpolate_q15'
sketch/DStarTX.cpp.o: In function `CDStarTX::writeByte(unsigned char)':
sketch/DStarTX.cpp:428: undefined reference to `arm_fir_interpolate_q15'
sketch/IO.cpp.o: In function `CIO::process()':
sketch/IO.cpp:340: undefined reference to `arm_q15_to_q31'
sketch/IO.cpp:343: undefined reference to `arm_biquad_cascade_df1_q31'
sketch/IO.cpp:361: undefined reference to `arm_fir_fast_q15'
sketch/IO.cpp:371: undefined reference to `arm_fir_fast_q15'
sketch/IO.cpp:381: undefined reference to `arm_fir_fast_q15'
sketch/IO.cpp:386: undefined reference to `arm_fir_fast_q15'
sketch/IO.cpp:393: undefined reference to `arm_fir_fast_q15'
sketch/IO.cpp.o:sketch/IO.cpp:409: more undefined references to `arm_fir_fast_q15' follow
sketch/NXDNTX.cpp.o: In function `CNXDNTX::writeByte(unsigned char)':
sketch/NXDNTX.cpp:151: undefined reference to `arm_fir_interpolate_q15'
sketch/NXDNTX.cpp:153: undefined reference to `arm_fir_fast_q15'
sketch/P25TX.cpp.o: In function `CP25TX::writeByte(unsigned char)':
sketch/P25TX.cpp:148: undefined reference to `arm_fir_interpolate_q15'
sketch/P25TX.cpp:150: undefined reference to `arm_fir_fast_q15'
sketch/POCSAGTX.cpp.o: In function `CPOCSAGTX::writeByte(unsigned char)':
sketch/POCSAGTX.cpp:132: undefined reference to `arm_fir_fast_q15'
sketch/YSFTX.cpp.o: In function `CYSFTX::writeByte(unsigned char)':
sketch/YSFTX.cpp:161: undefined reference to `arm_fir_interpolate_q15'
sketch/YSFTX.cpp.o: In function `CYSFTX::writeSilence()':
sketch/YSFTX.cpp:171: undefined reference to `arm_fir_interpolate_q15'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino Due (Programming Port).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Thanks for you help if you can.

73 F4IGV Yoann

g4klx, 2020-04-26:
You need to modify your Arduino compilation system. The details are in BUILD.txt in the MMDVM directory.

F4IGV, 2020-04-26:
Thanks a lot jonathan for your help.

after two days of hard work because I don't know anything about programming, ;-), the mmdvm F4IGV relay works very well on the brandmeist network and locally with Pi-star and the mmdvm firmware for arduino due. thank you for your amazing work.

73

F4IGV

iddq, 2020-06-09:
F4IGV, please close the issue if resolved.

closed 2020-06-09 by F4IGV