g4klx/MMDVM #261: Compile For STM32F446 Nucleo

M0XZS, 2020-05-11:
Hi,

Two issues I am having:

1. CooCox.org no longer exists. You have to hunt around to find a copy to use.
2. When compiling I get the following errors with no mode pins defined:

compile:
    [mkdir] Skipping C:\projects\MMDVM\mmdvm_pi_446\Debug\bin because it already exists.
    [mkdir] Skipping C:\projects\MMDVM\mmdvm_pi_446\Debug\obj because it already exists.
       [cc] Starting dependency analysis for 39 files.
       [cc] Parsing ..\..\..\Config.h
       [cc] Parsing ..\..\..\Globals.h
       [cc] Parsing ..\..\..\Utils.h
       [cc] Parsing ..\..\..\RSSIRB.h
       [cc] Parsing ..\..\..\STM32F4XX_Lib\Device\stm32f4xx.h
       [cc] Parsing ..\..\..\STM32F4XX_Lib\CMSIS\Include\core_cm4.h
       [cc] Parsing ..\..\..\STM32F4XX_Lib\CMSIS\Include\core_cmInstr.h
       [cc] Parsing ..\..\..\STM32F4XX_Lib\CMSIS\Include\core_cmFunc.h
       [cc] Parsing ..\..\..\STM32F4XX_Lib\CMSIS\Include\core_cmSimd.h
       [cc] Parsing ..\..\..\SampleRB.h
       [cc] Parsing ..\..\..\SerialPort.cpp
       [cc] Parsing ..\..\..\SerialRB.h
       [cc] 13 files are up to date.
       [cc] 26 files to be recompiled from dependency analysis.
       [cc] 27 total files to be compiled.
       [cc] arm-none-eabi-gcc -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -Wall -ffunction-sections -g -Os -c -DSTM32F4XX -DUSE_STDPERIPH_DRIVER -D__ASSEMBLY__ -DSUPPORT_CPLUSPLUS -D__FPU_USED -DSTM32F446xx -DHSE_VALUE=12000000 -DSTM32F4_PI -DMODE_LEDS -IC:\projects\MMDVM\STM32F4XX_Lib\STM32F4xx_StdPeriph_Driver -IC:\projects\MMDVM\STM32F4XX_Lib\Device -IC:\projects -IC:\projects\MMDVM\STM32F4XX_Lib -IC:\ -IC:\projects\MMDVM\STM32F4XX_Lib\STM32F4xx_StdPeriph_Driver\include -IC:\projects\MMDVM\STM32F4XX_Lib\CMSIS\Include -IC:\projects\MMDVM\STM32F4XX_Lib\CMSIS -IC:\projects\MMDVM C:\projects\MMDVM\DMRDMORX.cpp C:\projects\MMDVM\CalRSSI.cpp C:\projects\MMDVM\CalDMR.cpp C:\projects\MMDVM\CalDStarTX.cpp C:\projects\MMDVM\DMRRX.cpp C:\projects\MMDVM\P25TX.cpp C:\projects\MMDVM\IOSTM.cpp C:\projects\MMDVM\DStarTX.cpp C:\projects\MMDVM\CalFM.cpp C:\projects\MMDVM\YSFRX.cpp C:\projects\MMDVM\DMRDMOTX.cpp C:\projects\MMDVM\DMRTX.cpp C:\projects\MMDVM\CWIdTX.cpp C:\projects\MMDVM\SerialPort.cpp C:\projects\MMDVM\MMDVM.cpp C:\projects\MMDVM\SerialSTM.cpp C:\projects\MMDVM\CalDStarRX.cpp C:\projects\MMDVM\DMRIdleRX.cpp C:\projects\MMDVM\P25RX.cpp C:\projects\MMDVM\DMRSlotRX.cpp C:\projects\MMDVM\IOTeensy.cpp C:\projects\MMDVM\IO.cpp C:\projects\MMDVM\YSFTX.cpp C:\projects\MMDVM\IODue.cpp C:\projects\MMDVM\DStarRX.cpp C:\projects\MMDVM\SerialArduino.cpp C:\projects\MMDVM\DMRSlotType.cpp
       [cc] C:\projects\MMDVM\IOSTM.cpp: In member function 'void CIO::setFMInt(bool)':
       [cc] C:\projects\MMDVM\IOSTM.cpp:1451:18: error: 'PORT_FM' was not declared in this scope
       [cc]     GPIO_WriteBit(PORT_FM,  PIN_FM, on ? Bit_SET : Bit_RESET);
       [cc]                   ^~~~~~~
       [cc] C:\projects\MMDVM\IOSTM.cpp:1451:18: note: suggested alternative: 'PORT_DMR'
       [cc]     GPIO_WriteBit(PORT_FM,  PIN_FM, on ? Bit_SET : Bit_RESET);
       [cc]                   ^~~~~~~
       [cc]                   PORT_DMR
       [cc] C:\projects\MMDVM\IOSTM.cpp:1451:28: error: 'PIN_FM' was not declared in this scope
       [cc]     GPIO_WriteBit(PORT_FM,  PIN_FM, on ? Bit_SET : Bit_RESET);
       [cc]                             ^~~~~~
       [cc] C:\projects\MMDVM\IOSTM.cpp:1451:28: note: suggested alternative: 'PIN_RX'
       [cc]     GPIO_WriteBit(PORT_FM,  PIN_FM, on ? Bit_SET : Bit_RESET);
       [cc]                             ^~~~~~
       [cc]                             PIN_RX

BUILD FAILED
Total time: 8 seconds
Any ideas?

Andy

M0XZS, 2020-05-11:
I moved pine 1451 in IOSTM.cpp as follows:

#if defined(MODE_PINS) && defined(STM32F4_NUCLEO_MORPHO_HEADER) && (defined(STM32F4_NUCLEO) || defined(STM32F722_RPT_HAT))
   GPIO_WriteBit(PORT_FM,  PIN_FM, on ? Bit_SET : Bit_RESET);
   GPIO_WriteBit(PORT_MFM, PIN_MFM, on ? Bit_SET : Bit_RESET);
#endif

This now compiles but I get the following undefined references:

 arm-none-eabi-gcc -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -Wall -ffunction-sections -g -Os -c -DSTM32F4XX -DUSE_STDPERIPH_DRIVER -D__ASSEMBLY__ -DSUPPORT_CPLUSPLUS -DHSE_VALUE=8000000 -D__FPU_USED -DSTM32F4_NUCLEO -DSTM32F446xx -IC:\projects\MMDVM\STM32F4XX_Lib\STM32F4xx_StdPeriph_Driver -IC:\projects\MMDVM\STM32F4XX_Lib\Device -IC:\projects -IC:\projects\MMDVM\STM32F4XX_Lib -IC:\ -IC:\projects\MMDVM\STM32F4XX_Lib\STM32F4xx_StdPeriph_Driver\include -IC:\projects\MMDVM\STM32F4XX_Lib\CMSIS\Include -IC:\projects\MMDVM\STM32F4XX_Lib\CMSIS -IC:\projects\MMDVM C:\projects\MMDVM\IOSTM.cpp
       [cc] Starting link
       [cc] arm-none-eabi-gcc -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -Wall -g -nostartfiles -Wl,-Map=MMDVM_NUCLEO.map -Os -Wl,--gc-sections --specs=nano.specs -LC:/projects/MMDVM -Wl,-TC:/projects/MMDVM/stm32f4xx_link.ld -g -o MMDVM_NUCLEO.elf ..\obj\DMRDMORX.o ..\obj\RSSIRB.o ..\obj\CalRSSI.o ..\obj\CalDMR.o ..\obj\SampleRB.o ..\obj\CalDStarTX.o ..\obj\stm32f4xx_usart.o ..\obj\DMRRX.o ..\obj\P25TX.o ..\obj\IOSTM.o ..\obj\stm32f4xx_dac.o ..\obj\startup_stm32f4xx.o ..\obj\DStarTX.o ..\obj\CalFM.o ..\obj\stm32f4xx_rcc.o ..\obj\YSFRX.o ..\obj\stm32f4xx_adc.o ..\obj\DMRDMOTX.o ..\obj\DMRTX.o ..\obj\CWIdTX.o ..\obj\stm32f4xx_gpio.o ..\obj\SerialPort.o ..\obj\system_stm32f4xx.o ..\obj\MMDVM.o ..\obj\SerialSTM.o ..\obj\CalDStarRX.o ..\obj\DMRIdleRX.o ..\obj\Utils.o ..\obj\SerialRB.o ..\obj\P25RX.o ..\obj\DMRSlotRX.o ..\obj\IOTeensy.o ..\obj\IO.o ..\obj\YSFTX.o ..\obj\misc.o ..\obj\IODue.o ..\obj\DStarRX.o ..\obj\SerialArduino.o ..\obj\stm32f4xx_tim.o ..\obj\DMRSlotType.o -L..\..\..\STM32F4XX_Lib\CMSIS\Lib\GCC -larm_cortexm4lf_math -lm -lgcc -lc -lrdimon -lstdc++
       [cc] ..\obj\SerialPort.o: In function `CSerialPort::setFMParams3(unsigned char const*, unsigned char) [clone .part.2]':
       [cc] C:\projects\MMDVM/SerialPort.cpp:432: undefined reference to `CFM::setMisc(unsigned short, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, unsigned char, bool, bool, unsigned char, unsigned char, unsigned char)'
       [cc] C:\projects\MMDVM/SerialPort.cpp:232: undefined reference to `CNXDNTX::getSpace() const'
       [cc] C:\projects\MMDVM/SerialPort.cpp:237: undefined reference to `CPOCSAGTX::getSpace() const'
       [cc] ..\obj\SerialPort.o: In function `CSerialPort::setFMParams1(unsigned char const*, unsigned char)':
       [cc] ..\obj\SerialPort.o: In function `CSerialPort::getStatus()':
       [cc] C:\projects\MMDVM/SerialPort.cpp:387: undefined reference to `CFM::setCallsign(char const*, unsigned char, unsigned short, unsigned char, unsigned char, unsigned char, unsigned char, bool, bool, bool)'
       [cc] ..\obj\SerialPort.o: In function `CSerialPort::setFMParams2(unsigned char const*, unsigned char)':
       [cc] C:\projects\MMDVM/SerialPort.cpp:407: undefined reference to `CFM::setAck(char const*, unsigned char, unsigned short, unsigned char, unsigned short, unsigned char)'
       [cc] ..\obj\SerialPort.o: In function `CSerialPort::setMode(MMDVM_STATE)':
       [cc] C:\projects\MMDVM/SerialPort.cpp:554: undefined reference to `CNXDNRX::reset()'
       [cc] C:\projects\MMDVM/SerialPort.cpp:557: undefined reference to `CFM::reset()'
       [cc] ..\obj\SerialPort.o: In function `CSerialPort::setConfig(unsigned char const*, unsigned char)':
       [cc] C:\projects\MMDVM/SerialPort.cpp:348: undefined reference to `CPOCSAGTX::setTXDelay(unsigned char)'
       [cc] ..\obj\SerialPort.o: In function `CSerialPort::process()':
       [cc] C:\projects\MMDVM/SerialPort.cpp:347: undefined reference to `CNXDNTX::setTXDelay(unsigned char)'
       [cc] C:\projects\MMDVM/SerialPort.cpp:671: undefined reference to `CCalP25::write(unsigned char const*, unsigned char)'
       [cc] C:\projects\MMDVM/SerialPort.cpp:673: undefined reference to `CCalNXDN::write(unsigned char const*, unsigned char)'
       [cc] ..\obj\MMDVM.o: In function `loop()':
       [cc] C:\projects\MMDVM/SerialPort.cpp:854: undefined reference to `CNXDNTX::writeData(unsigned char const*, unsigned char)'
       [cc] C:\projects\MMDVM/MMDVM.cpp:108: undefined reference to `CNXDNTX::process()'
       [cc] C:\projects\MMDVM/SerialPort.cpp:675: undefined reference to `CCalPOCSAG::write(unsigned char const*, unsigned char)'
       [cc] C:\projects\MMDVM/SerialPort.cpp:868: undefined reference to `CPOCSAGTX::writeData(unsigned char const*, unsigned char)'
       [cc] C:\projects\MMDVM/MMDVM.cpp:111: undefined reference to `CPOCSAGTX::process()'
       [cc] C:\projects\MMDVM/MMDVM.cpp:114: undefined reference to `CFM::process()'
       [cc] C:\projects\MMDVM/MMDVM.cpp:123: undefined reference to `CCalP25::process()'
       [cc] C:\projects\MMDVM/MMDVM.cpp:126: undefined reference to `CCalNXDN::process()'
       [cc] C:\projects\MMDVM/MMDVM.cpp:129: undefined reference to `CCalPOCSAG::process()'
       [cc] C:\projects\MMDVM/MMDVM.cpp:110: undefined reference to `CPOCSAGTX::busy()'
       [cc] ..\obj\MMDVM.o: In function `__static_initialization_and_destruction_0':
[... cut ...]
       [cc] ..\obj\IO.o: In function `CIO::process()':
       [cc] C:\projects\MMDVM/IO.cpp:341: undefined reference to `CNXDNRX::samples(short const*, unsigned short*, unsigned char)'
       [cc] C:\projects\MMDVM/IO.cpp:362: undefined reference to `CFM::samples(bool, short*, unsigned char)'
       [cc] C:\projects\MMDVM/IO.cpp:423: undefined reference to `CNXDNRX::samples(short const*, unsigned short*, unsigned char
       [cc] )'
       [cc] collect2.exe: error: ld returned 1 exit status

BUILD FAILED
Total time: 2 seconds

Andy

shawnchain, 2020-05-21:
I tested with no error with following steps:
  * Checkout the latest code in master branch
  * enable either `#define STM32F4_NUCLEO_MORPHO_HEADER` or 
`#define STM32F4_NUCLEO_ARDUINO_HEADER` in config.h
  * make nucleo

```
arm-none-eabi-size bin/mmdvm_f4.elf
   text    data     bss     dec     hex filename
 286820    1684   52820  341324   5354c bin/mmdvm_f4.elf
arm-none-eabi-objcopy -O ihex bin/mmdvm_f4.elf bin/mmdvm_f4.hex
Objcopy from ELF to IHEX complete!

arm-none-eabi-objcopy -O binary bin/mmdvm_f4.elf bin/mmdvm_f4.bin
Objcopy from ELF to BINARY complete!
```

g4klx, 2020-05-21:
For some reason the mode LEDs work in some modes and not others.

g4klx, 2020-05-22:
Except when building for the ZUM v0,9 and v1.0 boards you use "build pi" and that doesn't set the LEDs correctly. I am sure that it used to. This is following the official instructions from Andy CA6JAU on mmdvm_man on GitHub.