g4klx/MMDVM #277: MMDMV_POG don't work anymore

badtime, 2020-05-16:
Hi,

since from commit 61ccd79c5f939a83c9380c2be3a0902b6b48117d the MMDVM_POG don't work anymore.
When power up it start the led staginng, but at half of staging it stop and stay with PTT and PWR led only powered UP.

c55002534e15a94cfad7a6c62da9ad46cb714bb2 is the lastest working commit

If i try to compile the lastest commit avalaible ff375abb0eb11cde7e56ac01ddb4b1957bc5de19 it is even worse, because when the POG power up, the led staging don't start and the board still dead with PWR led and the third led a little dimmed.

recapping:

Rename the Goertzel class to CTCSSRX. c55002534e15a94cfad7a6c62da9ad46cb714bb2 (lastest working)

Set up the FM mode pins, partially.  61ccd79c5f939a83c9380c2be3a0902b6b48117d (not work - half led staging)

Hold off the mode change until the ack and callsign have been ff375abb0eb11cde7e56ac01ddb4b1957bc5de19 (board dead, no staging)

What's happened?

73 IZ8GUR

yo6nam, 2020-05-22:
+1

I also reported this issue here :
https://groups.io/g/OpenDV/message/354

badtime, 2020-05-23:
> +1
> I also reported this issue here :
> https://groups.io/g/OpenDV/message/354

Yes, but the lastest working is 1332c6544b4bc4e8b6a0254b6d7fc0e4a1e3ee98 already in the "FM era".

but also the last 7fabddd7f15162e28704682aeaee7c6f93e328b8 sill not working, and the sad thing is that none of the developing ones are interested at now.

yo6nam, 2020-05-29:
Long story short, "POG" boards will not be supported anymore.
Have a read here : https://groups.io/g/OpenDV/message/367

badtime, 2020-05-29:
Yes....ok.....I can understand this in the near future...but...one commit of may 13 work on the POG (with FM ecc.ecc.), and in the same day another commit with a merge don't work......i think is only bad coding from merge request that does not consider niche boards IMHO.

yo6nam, 2020-05-29:
I don't really care about FM, although it would have been nice to make it work on repeaters already available, with just a FW upgrade. I just wish future development for the DMR will include support for the POG boards because I have plenty in use.

g4klx, 2020-05-29:
Please tell me which commit actually broke your board. I need the commit id, and I'll see what I can do.

g4klx, 2020-06-01:
I think it would also be very nice if SP6POG took responsibility for maintaining hos code. If I can't fix it, and if he doesn't step in to help, then I will delete the POG code from the MMDVM. I don't like people creating things that I can't maintain.

g4klx, 2020-06-01:
I've updated the master branch with an old version of the POG code from before the FM changes. I have no way of telling if its OK, but you can give it a try. If it doesn't work, contact SQ6POG.

badtime, 2020-06-01:
No way....dont work!
So, but....what's changed in "Merge branch 'master' into FM 8e52bdcc5c58cdf2e83b9319781ff49c2895c3f0" ?? Because after this merge of 9 MAY the POG stop Working...only 14 commits ago...12 if we remove the 2 of today. This is a quite recent trouble after that merge, and is in that merge which must be investigated IMHO.

g4klx, 2020-06-01:
I give up!
Can you fix it then? None of the main developers have such a board for testing and the developer seems to have disappeared. What are we to do?
I do this as a hobby and I don’t like it when people create non-standard hardware and expect us to keep it running. It isn’t fair.
Please try and contact SQ6POG and get him to fix it. If it’s not done in one week, all POG support will be removed from the MMDVM permanently. Do you understand?
Please do not demand that I do ANYTHING.

yo6nam, 2020-06-01:
Jonathan is right. Support for old hardware should be dropped (including Arduino which is also affected by the inclusion of FM code). The previous commits are stable enough to keep the POG boards in service with DMR/D-Star/C4FM.

Thanks for your time on this, Jonathan!

badtime, 2020-06-01:
Ok....i've saved the alstest working commit....I will continue to use it until the firmware is obsolete.

Thanks a lot Jonathan for the time the time you spend for us.

i will try to contact SP6POG hoping for the best.

closed 2020-06-01 by badtime

shawnchain, 2020-06-02:
I'm pretty sure FM mode requires more ram than a F103/105 can provide. Please switch to F4 with 128K+ ram or wait for other work around appears.

badtime, 2020-06-05:
Yes, it's true...but at this moment STM32F10X and the most powerful STM32F446 don't work anymore, both stop working after 1332c6544b4bc4e8b6a0254b6d7fc0e4a1e3ee98 so, I still don't think at now it's a performance issue.
At this point it would make sense to remove the support from all the other boards and leave only arduino two

> I'm pretty sure FM mode requires more ram than a F103/105 can provide. Please switch to F4 with 128K+ ram or wait for other work around appears.

SA7BNT, 2020-06-05:
> Yes, it's true...but at this moment STM32F10X and the most powerful STM32F446 don't work anymore, both stop working after [1332c65](https://github.com/g4klx/MMDVM/commit/1332c6544b4bc4e8b6a0254b6d7fc0e4a1e3ee98) so, I still don't think at now it's a performance issue.
> At this point it would make sense to remove the support from all the other boards and leave only arduino two
> 
 Just updated MMDVM Firmware. Using STM32F446 and Toufik modem Version 1.1
Seems to working well.

badtime, 2020-06-06:
What commit? 7fabddd7f15162e28704682aeaee7c6f93e328b8 or the last workaround 1 June?

> Just updated MMDVM Firmware. Using STM32F446 and Toufik modem Version 1.1
> Seems to working well.

scotta1980, 2020-06-20:
the sp8nth board no longer works also is this one of the bords that no longer will be supported