g4klx/MMDVM #283: Micro-Node Teensy not working after latest FW Upgrade

iansuart, 2020-05-30:
I have two units one with 2018 FW and one with 2019 FW. I decided they both needed upgrading so I started on the one with 2018 firmware.
Followed the instructions in the manual but noticed that the lines I should uncommunicative in config.h were not in the form shown in the manual.
I changed them to what I thought they should be an compiled
Compilation worked and upload to unit worked.
Pistar detects the modem firmware and the configured oscillator.
But it looks like mode data not being read to/from unit. No matter what I do in config.h makes no difference. Have I missed something?

iansuart, 2020-06-03:
This turned out to be an issue with MMDVMHost. Many many thanks to Andrew Taylor for sorting this out in Pi-Star

iddq, 2020-06-07:
What is the exact symptom? Could you control the tx with MMDVMCal?

mjswiatk, 2020-06-13:
Ian, can you post your config.h?  I have the same problem on the Micronode Teensy v3.6 also using PiStar.  I am racking my head on this.  What is the fix?  All I have is a flashing orange LED.

mjswiatk, 2020-06-13:
I found this thread: https://github.com/g4klx/MMDVM/issues/188
So, this is the latest version (4652812) that works, best I can tell.

iansuart, 2020-06-13:
Hi

I contacted the pi-star author and he worked through it and came up with a fix in MMDVMHost.

I had to do an update to get the fix

I can tell you my settings in config.h if you still need them

Ian

mjswiatk, 2020-06-13:
If you could attach your MMDVMHost file (or whatever else Andy found and fixed) that would be much appreciated.  This may also help the team here release a new version of MMDVM here that finally fixes the Teensy issues.

iansuart, 2020-06-14:
In config.h I ensured I had the the following lines uncommitted
30 - 12MHz oscillator
42 - mode pins
52 - modem type ZUM 1.0
69 - enable RSSI

In pi-star I selected the MMDVM/Teensy/ZUM. I did not use the MicroNide Teensy modem selection
MMDVMHost to follow

Ian

iansuart, 2020-06-14:
my MMDVCMHost

[mmdvmhost.txt](https://github.com/g4klx/MMDVM/files/4776363/mmdvmhost.txt)

mjswiatk, 2020-06-17:
I tried all of this with the latest Pi-Star build and MMDVM code.  Still, I can confirm there is an issue with Micronode Teensy MMDVM.  When powering  up I see 3/4 LEDs (no DStar).  I also see a blinking orange LED every 1s versus the quickly flashing Orange when the MMDVM firmware is properly loading on  the device.  I am reverting back to 4652812 until someone has a fix. Something is not right.

mjswiatk, 2020-06-26:
Has anyone had a chance to look at this issue yet?

g4klx, 2020-06-26:
It is up to your vendor to support you. None of the MMDVM developers have that hardware so you need to go back to the person who sold it to you.
They make money off our work, and we get nothing.

mjswiatk, 2020-06-26:
Jonathan,
I tried that.  No help.  I guess I will just run with 2 yr old code.  It's
the only thing that seems to work.

Mike

iansuart, 2020-06-26:
I am running with the latest code on two repeaters using micro ode Teensy 3.6 modems

Ian

mjswiatk, 2020-07-01:
No luck here.  I have the Micronode Teensy 3.6.  I followed your directions.  I don't think this issue has anything to do with PiStar.  The power up sequencing for the mode LEDs is not even correct, nor does it communicate with the Pi_Star.

My thought is that the config.h has changed.  For MicroNode Teensy they suggest commenting out these lines shown below, but as you can see they are renamed in the MMDVM code.  I suppose ARDUINO_MODE_PINS has become MODE_LEDS?  Could this be the problem?  MODE_PINS has an entirely different purpose.

#define EXTERNAL_OSC 12000000 
(Remove the // from the start of this line)

// Use pins to output the current mode layout
#define ARDUINO_MODE_PINS 
(Make sure the // is removed from this line as well)

// For the ZUM V1.0 and V1.0.1 boards pin 
#define ARDUINO_DUE_ZUM_V10 
(Make sure the // is removed from thisline as well)

mjswiatk, 2020-07-01:
I worked last night to try and find the most recent version without an issue on my teensy 3.6 MicroNode device.  I found that this version seems to be it:  https://github.com/g4klx/MMDVM/tree/b4647d8d1b10a60e7ca442f64a9c2ea3047dcee2

Some newer versions failed to verify when compiled or did not load on the device properly (i.e. LED sequencing indicating this  upon reboot).

mjswiatk, 2020-07-18:
Update:  I managed to get this version to work on Teensy 3.6:
4d664c4
on Apr 25, 2020

However, this version seems to be the latest to work on Teensy 3.2
78ae7f3
on Feb 9, 2018

It seems the best bet is to only use the "Verified" releases.  What is different between 3.2 and 3.6 in terms of firmware compatibility?

g4klx, 2020-07-20:
As I have said before, the Micro-Node stuff isn't supported by me. Pleas talk to your vendor, you payed them money for it, not me.

closed 2020-07-20 by g4klx