g4klx/MMDVM #303: Loss of connection to master server after updating mmdvm

DensRever, 2020-11-27:
Good morning all
Following a recent update of MMDVM I just lost the connection to the Brandmeister master.
The relay has since been in stand-alone mode and no longer makes it possible to receive and send voice to the Brandmeister TGs.
The MMDVM.ini is the same and the console no longer displays the fact that the connection is established to the master.
What procedure would allow me to return to normal?
Thank you
F4AJE Op Denis.

g7npw, 2020-11-27:
You now need to use the DMRGateway as well as your MMDVMHost to gain 
network connectivity to Brandmeister etc.

The MMDVHMost no longer connects direct.

Dom G7NPW

DensRever, 2020-11-28:
Hello Dom and thank you for your answer.
I am a bit of a novice. Can I update the existing version or do I have to start with a new version of MMDVM?
cordially
Denis