g4klx/MMDVM #312: Possible to add DCDM support?

jamesham, 2021-02-23:
Based on some discussion on the Pi-Star Facebook group, I started to wonder if MMDVM systems could support DMR Dual Capacity Direct Mode (DCDM). I don't think this is something MMDVM currently supports. Is there a reason it couldn't be added? Or just not popular enough to justify the development?

I saw that this was asked a few years ago, but the issue, #161, was closed without comment.

g4klx, 2021-03-01:
The MMDVM could do it easily, do you have any suitable radios with TX/RX changeover time of 1ms or less to do it with?

majkrzak, 2022-10-14:
Anytone 878 seems to support it. Can MMDCM on pistar can be configured to do it?

g4klx, 2022-10-14:
No. I have no programming information about it. If you find it, let me know.

RdWing, 2024-05-10:
TDMA direct mode (DCDM) is documented in the DMR standards documentation.
https://www.dmrassociation.org/dmr-standards.html Part 1, Sections 4.4.4, 4.6.2, 4.6.3, 5.1.4.6, 5.2.2.4, and more. I think TDMA direct mode is getting pretty popular, at least all the DMR Tier 2 radios I know of support it, including the ever popular Anytone 878, GD-77, DM-1701, MD-UV380/390, etc etc.

This could be a huge revolution for hotspots. If there are no other constraints, a "simplex" hotspot could function with two timeslots, as the air interface specs a bidirectional channel with guard time. I think it could be a great step forward, eliminating the need for and cost of two RF interfaces.

g4klx, 2024-05-10:
That's all very well, but can you find suitable hardware that can do the fast switching reliably?
Have a look at the timing requirements, they're very strict. It'd be very easy to do in software, but suitable hardware is lacking.

RdWing, 2024-05-10:
So we're getting closer to the edges of my knowledge here, but is it the timing aspect that disqualifies the AD7021/current MMDVM_HS boards?

What about the S2-LP with an STM32? I think there's a sufficient clock on that setup.

Or even this Auctus A6/A7/A8. It's the same integrated chip that powers the extremely low cost COTRE DMR portables, among others. The single integrated transceiver and controller must be very cheap, and I can confirm it is capable of DCDM. If I can get a data sheet I'll return with that.