g4klx/MMDVM #315: RTOS feasibility

nbgsmk, 2021-03-18:
Hello Jonathan, and everyone else involved!
Thanks for the great work. What you made has truly impacted the whole world, at least in ham radio circles. Not many people claim that, in any field. 

This is not a real issue, but a query out of my own curosity as a developer. 
Could you explain why, for example FreeRTOS (or any other) was not used in this project? Involved processing is quite complex and using an rtos would bring a lot of advantages. On the other hand, problems like these immediately come to mind:
- was this just a design choice at the time
- lack of hard real-time capabilities needed for timeslot capture and processing
- avoiding additional load on the limited stm32 resources

Or am I completely missing the point? :-)
Thanks!

g4klx, 2021-03-22:
It was a design choice. The project first started on an 85 MHz ARM-M3 although most systems are more powerful these days. I did consider an RTOS for a short while, but what would it give me? I could find no advantage to using one, and only disadvantages. As it is, on DMR the system synchronises to within around 40 microseconds between TX and RX, and I fail to see how an RTOS would help me achieve this. On an embedded platform, I have not seen any overwhelming justification for using an RTOS. As it is, the MMDVM firmware supports Atmel, NXP, and ST-Micro ARM chips without problems.

nbgsmk, 2021-03-22:
Okay Andy, thanks a lot! It was just a question.

closed 2021-03-22 by nbgsmk