g4klx/MMDVM #319: FM DAC Overflow

G0FHM, 2021-09-14:
I'm using a Winters (BI7JTA) V3F4 repeater board with pistar 4,1,5 on a Pi 3B+

I have an issue whereby when FM is enabled and the repeater is open, i'm sometimes getting it being held open by near constant kerchunking to the point where it will time out, or I have to reboot the Pi to stop it..

Looking at the live logs, pi-star is seeing it as a valid signal with CTCSS each kerchunk, but the live log constantly scrolls with "MMDVM DAC levels have overflowed"

The board is using the VR2VYE firmware 20201214.

2x  Kenwood TK-8180, 6 Pye cavities, 70db+ isolation.

Would this be a firmware issue?

Thanks,
Jay G0FHM (GB3GS)

m0vse, 2021-11-18:
I would try reducing the FM TX level within MMDVM.ini as 'DAC levels have overflowed' means that the TX audio level is higher than it can handle. 

Also, that's quite an old version as there have been many fixes in the last 11 months!

73 Phil M0VSE

bi7jta, 2024-04-14:
Solve "MMDVM DAC levels have overflowed" problem, see my WIKI guide.

Turn the RX pot clockwise until the CLIP LED turns off

https://www.bi7jta.org/wiki/index.php?title=Main_Page#The_situation_must_adjust_RXLevel

[image]

m0llc, 2024-04-16:
I have this problem too. I dont get the DAC error just a random TX and noise after some overs. I am using GM340 and Winters (BI7JTA) V3F4 repeater board with JSTAR. The error logs point to CTCSS Opening

kandev, 2025-05-31:
Did some of you guys found a solution for this?
I'm using pair of GM340s with BI7JTA'a MMDVM controller and in FM mode I get dirty tail from time to time and "MMDVM DAC levels have overflowed" in the logs.