g4klx/MMDVM #328: Switching channel when pocsag message is sent

pocsag, 2022-09-20:
Dear all,

I would like to switch my motorola GM360 to 439,9875 when a pocsag message is beeing sent.
https://www.florian-wolters.de/posts/automated-channel-switching-with-motorola-gm3x0/
I use the pocsag led on my repeaterboard to switch the frequency.
It seems the pocsag led is beeing switched at the same time as the PTT turns on.
Because the motorola is in TX, the frequency will not switch anymore.
Is there another gpio useable when a pocsag message is sent?
Or maybe is the PTT delayable?
I hope someone can help me out. Thanks in advance!
Greets, Joep PD1AEF.

capnjb, 2022-09-21:
Joep,
You may be able to accomplish what you want to do by using the Emergency function/switch (Pin 9?).  This should QSY to the programmed Emergency channel.  I am not sure exactly on the timing that you have set up but it may work for you.

Good Luck,
73
Jon
K1IMD

> Dear all,
> 
> I would like to switch my motorola GM360 to 439,9875 when a pocsag message is beeing sent. .

phl0, 2022-09-21:
To be honest this never really worked good and was more like a PoC. I also found that timing is an issue and especially with duplex repeaters it did not really work well.

pocsag, 2022-09-21:
Hi all,
The switching works great.
If the pocsag LED goes high it switches.
If only PTT would kick in a few mili seconds later it would be okay...
I was hoping to find out where the code could be altered for this.
Regards, Joep

mytechguyri, 2023-01-26:
Barring a change in code I suppose one solution would be route PTT and the LED into a microcontroller... then you could program a few millisecond delay after PTT is asserted before passing the PTT signal through to the radio to give it enough time to change channel.