g4klx/MMDVM #337: pls bring back support for STM32F105/POG

invictus737, 2023-09-25:
why drop support for POG STM32F105 in recent code? It's OK to have limited features but up to date code is important for old boards.

Thx!

g4klx, 2023-09-25:
I didn’t realise that it’d stopped working! I would not recommend such a board these days, too underpowered. How do you build the firmware, so I can investigate what has gone wrong.

invictus737, 2023-09-25:
Hi Jonathan,

regular build with minimum activated modes. DMR YSF and M17(forced to enable as i get errors if i comment M17).
my order of commands:
git clone https://github.com/g4klx/MMDVM.git
cd MMDVM
git submodule init
git submodule update
nano Config.h https://pastebin.com/U4fTRnQT
make -f Makefile.CMSIS https://pastebin.com/DwRT87vY
build is OK no errors just a warning and then stlinkv2 flash and board boots but not detected at 115200 with STM32-DV gpio hat settings with W0CHP pistar image. old build from 2019 works ok

g4klx, 2023-09-25:
Hi
I can't really answer your issue, but it may be worth trying the higher serial speeds, in particular 460800, in case it helps. Otherwise I have no clue as to how to fix your problem. I have no similar hardware here to test on, and as I said I cannot recommend such hardware these days. If the worst comes to the worst stay with an older firmware version.
You *may* want to try commenting out the 460800 speed higher up in the config file and uncommenting the 115200 speed line.
I strongly advise upgrading to something more capable. If the problem still occurs then I'll seriously consider removing support for this older board as it was always underpowered even when new. SP6POG released his hardware without consultation with me or anyone else, and we would have advised against it even in 2016.
Jonathan  G4KLX

invictus737, 2023-09-25:
Hi Jonathan,

tried everything by now... lower and higher speeds, more and few modes, nothing. Current code is not compatible with STM32F105. That is strange as MMDVM-HS(fork) boots just fine for jumbo/zum simplex/duplex hotspots with STM32F103 :)))
Would be nice to have at least for old repeater boards DMR and FM mode working... I'm sure it can be done if someone can identify/debug when&why the code broke STM32F105 POG during the timeline.

Many thanks for all your current efforts!

Regards,
Chris/YO3TCO

invictus737, 2023-09-26:
I've search back and found references of main branch not working since 07.2021 when compiled for STM32F1
Working firmware from my logs is: MMDVM protocol version: 1, description: MMDVM 20190130 (D-Star/DMR/System Fusion/P25/NXDN/POCSAG) 19.2000 Mhz GitID #b4647d8 
https://github.com/g4klx/MMDVM/commit/b4647d8d1b10a60e7ca442f64a9c2ea3047dcee2 is the working github commit.

Regards,
Chris/YO3TCO

ct1jib, 2023-09-27:
> There is no way that the old board will work on FM, it requires too much
> processing. I cannot see any reason for the 105 based boards not to work,
> so I assume that it is now too big for the small processor, that allied
> with the lack of processing on it, make it unsuitable for use with modern
> firmware versions. If you want to try modern modes and options, you have to
> use suitable modern hardware.
>
> I will remove 105 support from the MMDVM firmware as it is no longer
> suitable for running it. Older versions of the firmware will remain
> available but will receive no updates.

Why? Why a system named multi digital voice goes to analogic and this is
more important that the project itself? Not understand...!

donwade, 2023-12-25:
Jonathan.
Can you give me advice what modem board you would recommend given your opinion of the 103? My old dual hat version is a 103 (got if from walmart at $33). Also what do you use for a host  USB on a PC, PI?

Just trying to get started on this project and would like to hit the ground running by aligning with you.

invictus737, 2023-12-26:
This thread was for the repeater boards. Your duplex hotspot is just fine with STM32F103 using this firmware. https://github.com/juribeparada/MMDVM_HSRegardsChris

g4klx, 2023-12-26:
You should be using the MMDVM_HS firmware from my GitHub.

donwade, 2023-12-26:
I'm aware of the MMDVM_HS link that supports the 103 and I've built and heavily modified it. Upon doing a diff b/n that repository and this one, I find this one has way more interesting functionality that that a 103 can't deliver, like CTSS etc.
So should have said I want to ditch my 103 and move to whatever platform you use for repeater boards (interested in ADC/D2A aspects for different reasons).
I'm not a ham person but do have 30 years embedded / radio experience. I tried to search the internet for what radio board you could be using, but everything brought me back to 103/105 solutions. What boards use the STMF4/F7/or-better series? Can you provide me any links?

closed 2024-09-07 by g4klx