g4klx/MMDVM #339: MMDVM stuck in FM mode on mqtt branch

ke5gdb, 2023-11-17:
This line causes a feedback loop with MMDVMHost. CFM::reset() is called when MMDVMHost asserts MMDVM_SET_MODE to IDLE, but serial.writeFMStatus() will cause MMDVMHost to assert the mode back into FM. Ultimately this locks the modem in FM mode.

https://github.com/g4klx/MMDVM/blame/a8bf99037ec31546321a6fc69e355f77d9d1aaef/FM.cpp#L391

I commented on the line here in the commit, but this comment is likely not the correct medium, hence the new issue.

https://github.com/g4klx/MMDVM/commit/7ef0d992d9190466bf13b7306b8867358898c9f1#r132809346

73 de KE5GDB

g4klx, 2023-11-18:
Hi Andrew
I think your fix is correct. I've incorporated it into the mqtt branch of the MMDVM firmware and I hope that the problem has now gone away. Thanks for finding it.
Jonathan  G4KLX

 
 

https://github.com/g4klx/MMDVM/blame/a8bf99037ec31546321a6fc69e355f77d9d1aaef/FM.cpp#L391

7ef0d99#r132809346

bi7jta, 2024-01-12:
Hi @g4klx Jonathan ,
I got this problem "stuck in FM mode" when I Not set the configuration in mmdvmhost,

`[FM]
...
ModeHang=10 `

Do you think add  is a good idea to add default ModeHang (Not Infinity), in firmware or in MMDVMHost?

Thank you very much!

DE BI7JTA

g4klx, 2024-01-12:
Can you try the latest MMDVM firmware, this issue is fixed with it.

closed 2024-09-07 by g4klx