g4klx/MMDVM #341: Building FW

toyotahead, 2024-01-07:
My setup:
- RPI 3B
- Pi-Star:4.1.7 / Dashboard: 2024010  (clean and fresh install)
- ZUM Radio-MMDVM for Pi v1.0 (HRO white PCB) (presently has as purchased FW: MMDVM:20190130)
- 64GB new SD card (full size confirmed with df -h)
- Utilizing ethernet cable not wifi
- SSH'd into the pi via Putty
- instructions: https://www.hamradio.com/images_managed/misc/H0-016486_MMDVM_Pi_1_User_Manual_1_2020.pdf (in the pi-star sections)

Goal:
Update FW to enable FM & M17 modes

I have done this procedure in the past with a different board without issue. However this time I am being hung up at the compiling stage. I am able to get to the "make pi-f722" where it fails with:

```
pi-star@pi-star(rw):MMDVM$ make pi-f722
arm-none-eabi-g++ -Os -fno-exceptions -ffunction-sections -fdata-sections -fno-builtin -fno-rtti -DCUSTOM_NEW -DNO_EXCEPTIONS -c -mcpu=cortex-m7 -mthumb -mlittle-endian -mfpu=fpv5-sp-d16 -mfloat-abi=hard -mthumb-interwork -I. -I./STM32F7XX_Lib/CMSIS/Include/ -I./STM32F7XX_Lib/Device/ -I./STM32F7XX_Lib/STM32F7xx_StdPeriph_Driver/inc/ -DUSE_HAL_DRIVER -DSTM32F722xx -DSTM32F7XX -DSTM32F722_PI -DHSE_VALUE=12000000 -DMADEBYMAKEFILE SerialSTM.cpp -o obj_f7/SerialSTM.o
In file included from SerialSTM.cpp:21:0:
Config.h:26:1: error: 'define' does not name a type
 define MODE_DSTAR
 ^~~~~~
In file included from ./STM32F7XX_Lib/CMSIS/Include/core_cmInstr.h:61:0,
                 from ./STM32F7XX_Lib/CMSIS/Include/core_cm7.h:210,
                 from ./STM32F7XX_Lib/Device/stm32f722xx.h:180,
                 from ./STM32F7XX_Lib/Device/stm32f7xx.h:132,
                 from Globals.h:25,
                 from SerialSTM.cpp:22:
./STM32F7XX_Lib/CMSIS/Include/cmsis_gcc.h:40:28: error: expected declaration before end of line

                            ^
make: *** [Makefile:312: obj_f7/SerialSTM.o] Error 1

```

My Config.h file:
```
/*
 *   Copyright (C) 2015,2016,2017,2018,2020 by Jonathan Naylor G4KLX
 *
 *   This program is free software; you can redistribute it and/or modify
 *   it under the terms of the GNU General Public License as published by
 *   the Free Software Foundation; either version 2 of the License, or
 *   (at your option) any later version.
 *
 *   This program is distributed in the hope that it will be useful,
 *   but WITHOUT ANY WARRANTY; without even the implied warranty of
 *   MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 *   GNU General Public License for more details.
 *
 *   You should have received a copy of the GNU General Public License
 *   along with this program; if not, write to the Free Software
 *   Foundation, Inc., 675 Mass Ave, Cambridge, MA 02139, USA.
 */

#if !defined(CONFIG_H)
#define  CONFIG_H

// Allow for the selection of which modes to compile into the firmware. This is particularly useful for processors
// which have limited code space and processing power.

// Enable D-Star support.
define MODE_DSTAR

// Enable DMR support.
define MODE_DMR

// Enable System Fusion support.
define MODE_YSF

// Enable P25 phase 1 support.
define MODE_P25

// Enable NXDN support, the boxcar filter sometimes improves the performance of NXDN receive on some systems.
define MODE_NXDN
define USE_NXDN_BOXCAR

// Enable M17 support.
define MODE_M17

// Enable POCSAG support.
define MODE_POCSAG

// Enable FM support.
define MODE_FM

// Enable AX.25 support, this is only enabled if FM is also enabled.
define MODE_AX25

// Allow for the use of high quality external clock oscillators
// The number is the frequency of the oscillator in Hertz.
//
// The frequency of the TCXO must be an integer multiple of 48000.
// Frequencies such as 12.0 Mhz (48000 * 250) and 14.4 Mhz (48000 * 300) are suitable.
// Frequencies such as 10.0 Mhz (48000 * 208.333) or 20 Mhz (48000 * 416.666) are not suitable.
//
// For 12 MHz
#define EXTERNAL_OSC 12000000
// For 12.288 MHz
// #define EXTERNAL_OSC 12288000
// For 14.4 MHz
// #define EXTERNAL_OSC 14400000
// For 19.2 MHz
// #define EXTERNAL_OSC 19200000

// Select a baud rate for host communication. The faster speeds are needed for external FM to work.
// #define SERIAL_SPEED 115200 // Suitable for most older boards (Arduino Due, etc). External FM will NOT work with this!
// #define SERIAL_SPEED 230400 // Only works on newer boards like fast M4, M7, Teensy 3.x. External FM might work with this
define SERIAL_SPEED 460800	// Only works on newer boards like fast M4, M7, Teensy 3.x. External FM should work with this
//#define SERIAL_SPEED 500000  // Used with newer boards and Armbian on AllWinner SOCs (H2, H3) that do not support 460800

// Use pins to output the current mode via LEDs
define MODE_LEDS

// For the original Arduino Due pin layout
// #define ARDUINO_DUE_PAPA

// For the ZUM V1.0 and V1.0.1 boards pin layout
// #define ARDUINO_DUE_ZUM_V10

// For the SP8NTH board
// #define ARDUINO_DUE_NTH

// For ST Nucleo-64 STM32F446RE board
// #define STM32F4_NUCLEO_MORPHO_HEADER
// #define STM32F4_NUCLEO_ARDUINO_HEADER

// Use separate mode pins to switch external channel/filters/bandwidth for example
// #define MODE_PINS

// For the VK6MST Pi3 Shield communicating over i2c. i2c address & speed defined in i2cTeensy.cpp
// #define VK6MST_TEENSY_PI3_SHIELD_I2C

// Pass RSSI information to the host
define SEND_RSSI_DATA

// Use the modem as a serial repeater for Nextion displays
define SERIAL_REPEATER

// Use the modem as an I2C repeater for OLED displays
// #define I2C_REPEATER

// To reduce CPU load, you can remove the DC blocker by commenting out the next line
define USE_DCBLOCKER

// Constant Service LED once repeater is running 
// Do not use if employing an external hardware watchdog 
// #define CONSTANT_SRV_LED

// Use the YSF and P25 LEDs for NXDN
// #define USE_ALTERNATE_NXDN_LEDS

// Use the D-Star and P25 LEDs for M17
#define USE_ALTERNATE_M17_LEDS

// Use the D-Star and DMR LEDs for POCSAG
#define USE_ALTERNATE_POCSAG_LEDS

// Use the D-Star and YSF LEDs for FM
#define USE_ALTERNATE_FM_LEDS

#endif
```

It doesn't seem to matter what is uncommented, these define error follows the first line that is uncommented. 

What have I missed doing to progress this compiling process?

m0vse, 2024-01-11:
You have removed the # from in-front of all of the define commands, this is part of the command and NOT a comment!
They should be '#define xxxxx'

Phil

toyotahead, 2024-01-11:
Thank you Phil

So I am not quite sure why the instruction says to "uncomment" certain lines as the default Config.h looks like you are suggesting. Perhaps I just didn't understand....

"#define xxxxx'     (minus the single quotes)

If I am to understand you correctly I just need to put back all the # and then run the 'make pi-f722' command and all these feature will be enabled by default? To which I have tried and indeed I now have files populated in my /bin folder. YA!!! (hopefully the features are enabled :)  )

I continued with the instructions into the flashing section. Everything seems fine right up to where it begins to flash where errors are encountered (see below) and the mmdvm board seems to lock up with a solid red PTT light. A power cycle resolves this and both the pi and the mmdvm start up as per normal with its old firmware 20190130

The specific command within the flashing script file that seems to fail looks like: sudo stm32flash -v -w mmdvm_f7.bin -g 0x0 -R -c /dev/ttyAMA0
I have read somewhere it is the -c switch that causes problems but I am hesitant to just try removing it without seeking assurance I am on the right track. Are you able to advise further?

73 Thanks again!
PU3ZJM
VE7TEQ

`
pi-star@pi-star(rw):bin$ ./MMDVM-Pi_FW_Update.sh
+ sudo mount -o remount,rw /
+ sudo mount -o remount,rw /boot
+ sudo apt-get remove libi2c-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'libi2c-dev' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
+ sudo apt-get remove stm32flash
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'stm32flash' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
+ sudo git clone https://git.code.sf.net/p/stm32flash/code stm32flash
Cloning into 'stm32flash'...
remote: Enumerating objects: 1530, done.
remote: Counting objects: 100% (1530/1530), done.
remote: Compressing objects: 100% (855/855), done.
remote: Total 1530 (delta 1031), reused 994 (delta 671)
Receiving objects: 100% (1530/1530), 1.07 MiB | 778.00 KiB/s, done.
Resolving deltas: 100% (1031/1031), done.
+ cd stm32flash
+ sudo make
cc -Wall -g   -c -o dev_table.o dev_table.c
cc -Wall -g   -c -o i2c.o i2c.c
cc -Wall -g   -c -o init.o init.c
cc -Wall -g   -c -o main.o main.c
cc -Wall -g   -c -o port.o port.c
cc -Wall -g   -c -o serial_common.o serial_common.c
cc -Wall -g   -c -o serial_platform.o serial_platform.c
cc -Wall -g   -c -o stm32.o stm32.c
cc -Wall -g   -c -o utils.o utils.c
cd parsers && make parsers.a
make[1]: Entering directory '/home/pi-star/MMDVM/bin/stm32flash/parsers'
cc -Wall -g   -c -o binary.o binary.c
cc -Wall -g   -c -o hex.o hex.c
ar rc parsers.a binary.o hex.o
make[1]: Leaving directory '/home/pi-star/MMDVM/bin/stm32flash/parsers'
cc  -o stm32flash dev_table.o i2c.o init.o main.o port.o serial_common.o serial_platform.o stm32.o utils.o parsers/parsers.a
+ sudo make install
cd parsers && make parsers.a
make[1]: Entering directory '/home/pi-star/MMDVM/bin/stm32flash/parsers'
make[1]: 'parsers.a' is up to date.
make[1]: Leaving directory '/home/pi-star/MMDVM/bin/stm32flash/parsers'
install -d /usr/local/bin
install -m 755 stm32flash /usr/local/bin
install -d /usr/local/share/man/man1
install -m 644 stm32flash.1 /usr/local/share/man/man1
+ cd ../
+ sudo pistar-watchdog.service stop
Killing pistar-watchdog PID 1556
+ sudo systemctl stop mmdvmhost.timer
+ sudo systemctl stop mmdvmhost.service
+ sudo stm32flash -i 20,-21,21:-20,21 /dev/ttyAMA0
stm32flash 0.7

http://stm32flash.sourceforge.net/

Interface serial_posix: 57600 8E1

GPIO sequence start
 setting gpio 20 to 1... OK
 delay 100000 us
 setting gpio 21 to 0... OK
 delay 100000 us
 setting gpio 21 to 1... OK
GPIO sequence end

Failed to init device.

GPIO sequence start
 setting gpio 20 to 0... OK
 delay 100000 us
 setting gpio 21 to 1... OK
GPIO sequence end

+ sudo stm32flash -v -w mmdvm_f7.bin -g 0x0 -R -c /dev/ttyAMA0
stm32flash 0.7

http://stm32flash.sourceforge.net/

Using Parser : Raw BINARY
Size         : 340108
Interface serial_posix: 57600 8E1
Failed to read ACK byte
Unexpected reply from device on command 0x01

+ sudo reboot now

Remote side unexpectedly closed network connection
`

toyotahead, 2024-01-14:
sad news...  tried with the -c removed. Flash appeared to work as it was reporting "Wrote and verified 100%" but upon reboot only the power led is illuminated and no flashing STAT led. Upon checking pi-star dashboard no modem is recognized / attached. 

Thinking the fw didn't flash I ran the same exact procedure again. Once again I see the PTT led illuminate and the fw begins and reports "Wrote and verified 100%". Once complete the PTT led extinguishes leaving only the PWR led on.

Rebooted

Same result, only the PWR led is on and no modem detected. I assume the flash did not work / corrupted the fw on the chip. But it seems as though that the chip is still present and responding as the ptt led going on and off. 

Also if I put the -c back into the flashing command I get the previous flashing error as I did before "Failed to read ACK byte" and the PTT and PWR leds are illuminated.

What is the path forward from here? Really hope this board isn't junk at this point :/

nimdaxunil, 2026-08-27:
> sad news... tried with the -c removed. Flash appeared to work as it was reporting "Wrote and verified 100%" but upon reboot only the power led is illuminated and no flashing STAT led. Upon checking [...]

Curious - was a solution ever found for this particular issue? I am having exactly the same result. Any update would be appreciated - thank you.

nimdaxunil, 2026-08-27:
> Thank you Phil
> 
> Not sure if it is relevant but, I forgot to mention this MMDVM board is bran new. It has never been used before.
> 
> So I am not quite sure why the instruction says to "uncomment" certain lines as the default Config.h looks like you are suggesting. Perhaps I just didn't understand....
> 
> "#define xxxxx' (minus the single quotes)
> 
> If I am to understand you correctly I just need to put back all the # and then run the 'make pi-f722' command and all these feature will be enabled by default? To which I have tried and indeed I now have files populated in my /bin folder. YA!!! (hopefully the features are enabled :) )
> 
> I continued with the instructions into the flashing section. Everything seems fine right up to where it begins to flash where errors are encountered (see below) and the mmdvm board seems to lock up with a solid red PTT, and solid orange PWR led lights only. (no rapid flashing STAT anymore). A power cycle resolves this and both the pi and the mmdvm start up as per normal with its old firmware 20190130
> 
> The specific command within the flashing script file that seems to fail looks like: sudo stm32flash -v -w mmdvm_f7.bin -g 0x0 -R -c /dev/ttyAMA0 I have read somewhere it is the -c switch that causes problems but I am hesitant to just try removing it without seeking assurance I am on the right track. Are you able to advise further?
> 
> 73 Thanks again! PU3ZJM VE7TEQ
> 
> ```
> pi-star@pi-star(rw):bin$ ./MMDVM-Pi_FW_Update.sh
> + sudo mount -o remount,rw /
> + sudo mount -o remount,rw /boot
> + sudo apt-get remove libi2c-dev
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> Package 'libi2c-dev' is not installed, so not removed
> 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
> + sudo apt-get remove stm32flash
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> Package 'stm32flash' is not installed, so not removed
> 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
> + sudo git clone https://git.code.sf.net/p/stm32flash/code stm32flash
> Cloning into 'stm32flash'...
> remote: Enumerating objects: 1530, done.
> remote: Counting objects: 100% (1530/1530), done.
> remote: Compressing objects: 100% (855/855), done.
> remote: Total 1530 (delta 1031), reused 994 (delta 671)
> Receiving objects: 100% (1530/1530), 1.07 MiB | 778.00 KiB/s, done.
> Resolving deltas: 100% (1031/1031), done.
> + cd stm32flash
> + sudo make
> cc -Wall -g   -c -o dev_table.o dev_table.c
> cc -Wall -g   -c -o i2c.o i2c.c
> cc -Wall -g   -c -o init.o init.c
> cc -Wall -g   -c -o main.o main.c
> cc -Wall -g   -c -o port.o port.c
> cc -Wall -g   -c -o serial_common.o serial_common.c
> cc -Wall -g   -c -o serial_platform.o serial_platform.c
> cc -Wall -g   -c -o stm32.o stm32.c
> cc -Wall -g   -c -o utils.o utils.c
> cd parsers && make parsers.a
> make[1]: Entering directory '/home/pi-star/MMDVM/bin/stm32flash/parsers'
> cc -Wall -g   -c -o binary.o binary.c
> cc -Wall -g   -c -o hex.o hex.c
> ar rc parsers.a binary.o hex.o
> make[1]: Leaving directory '/home/pi-star/MMDVM/bin/stm32flash/parsers'
> cc  -o stm32flash dev_table.o i2c.o init.o main.o port.o serial_common.o serial_platform.o stm32.o utils.o parsers/parsers.a
> + sudo make install
> cd parsers && make parsers.a
> make[1]: Entering directory '/home/pi-star/MMDVM/bin/stm32flash/parsers'
> make[1]: 'parsers.a' is up to date.
> make[1]: Leaving directory '/home/pi-star/MMDVM/bin/stm32flash/parsers'
> install -d /usr/local/bin
> install -m 755 stm32flash /usr/local/bin
> install -d /usr/local/share/man/man1
> install -m 644 stm32flash.1 /usr/local/share/man/man1
> + cd ../
> + sudo pistar-watchdog.service stop
> Killing pistar-watchdog PID 1556
> + sudo systemctl stop mmdvmhost.timer
> + sudo systemctl stop mmdvmhost.service
> + sudo stm32flash -i 20,-21,21:-20,21 /dev/ttyAMA0
> stm32flash 0.7
> 
> http://stm32flash.sourceforge.net/
> 
> Interface serial_posix: 57600 8E1
> 
> GPIO sequence start
>  setting gpio 20 to 1... OK
>  delay 100000 us
>  setting gpio 21 to 0... OK
>  delay 100000 us
>  setting gpio 21 to 1... OK
> GPIO sequence end
> 
> Failed to init device.
> 
> GPIO sequence start
>  setting gpio 20 to 0... OK
>  delay 100000 us
>  setting gpio 21 to 1... OK
> GPIO sequence end
> 
> 
> + sudo stm32flash -v -w mmdvm_f7.bin -g 0x0 -R -c /dev/ttyAMA0
> stm32flash 0.7
> 
> http://stm32flash.sourceforge.net/
> 
> Using Parser : Raw BINARY
> Size         : 340108
> Interface serial_posix: 57600 8E1
> Failed to read ACK byte
> Unexpected reply from device on command 0x01
> 
> + sudo reboot now
> 
> Remote side unexpectedly closed network connection
> ```

Curious - was a solution ever found for this particular issue? I am having exactly the same result. Any update would be appreciated - thank you.