g4klx/MMDVM #348: Send POCSAG from ESP32?

smithse, 2025-04-11:
This is not an issue, but a question around  the use of an ESP32 board to control the MMDVM.

I have a spare simplex hat that I am toying with the idea of turning into a POCSAG transmitter. I was hoping to use an ESP32 board dev board to host a http api that I could use to send pager messages locally.

In short, would this be possible?

g4klx, 2025-04-11:
The easiest way is to inject commands into the MMDVM Host, using the remote control interface, it does all of the work for you. If you want to go directly to the MMDVM hat then you need to read and understand what is done by the host to format the messages for the hat. I can't remember how I did it, I wrote the code seven years ago and I haven't needed to modify it since.

smithse, 2025-04-14:
Thanks for the info Jonathan! I'm currently using the remote commands on my pistar and it works great. Using the mmdvm board in a standalone setup is more of a challenge that I have set myself. I'll dig in and see what I can work out! 
Cheers